OrcaSlicer/OrcaSlicer #3905: Bridges classified as bottom surface when supports enabled and top distance is 0

gshpychka, 2024-01-30:
### OrcaSlicer Version

1.9.0

### OS version

macOS 14.2.1

### Printer

Bambu Lab P1S

### How to reproduce

I have tried to isolate the issue, but I am not confident about the exact steps to take.

1. Have a model with an internal bridge that has full layers below it
2. Enable supports, but only on the build plate (so that the bridge is not supported)
3. Set the top Z distance to 0
4. I have also seen it happen when disabling "don't support bridges", but cannot reproduce consistently

### Actual results

The bridge becomes "Bottom surface", printed fast (with the same speed as "First layer infill") and diagonally
[image]

### Expected results

The bridge is classified as "Internal bridge", printed slowly and perpendicular.

Project file attached.

[image]

### Project file & Debug log uploads

[bug_repro_2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14099214/bug_repro_2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

enz1ey, 2024-04-12:
This still seems to be an issue as of 1.9.1. After lots of tinkering, the best workaround I found was to simply change the "Top Z distance" value to a decimal value so close to 0 the slicer wouldn't actually change anything with the layers. I used a value of "0.0001" and since that value wasn't exactly 0, the line type did in fact change from "bottom surface" to "bridge" and there was no discernible offset between the support interface and next layer. While the line type still wasn't generating as "internal bridge," it still solved my issue for now.

gshpychka, 2024-07-12:
This is still relevant.

enz1ey, 2024-07-12:
I concur with @gshpychka this is still happening as of v2.1.1. See photos below.

Layer preceding support layer for reference:
[image]

**Support** - **Advanced** - **Top Z distance** set to 0:
Support layer:
[image]

**Support** - **Advanced** - **Top Z distance** set to 0.001:
[image]

gshpychka, 2024-10-11:
The issue is still relevant

gshpychka, 2025-01-10:
Still relevant.

gshpychka, 2025-04-11:
Still relevant.

Felix14-v2, 2026-10-06:
> 3\. Set the top Z distance to 0

That's the root of the problem. Slicer interprets all bridges as bottom surfaces if the distance is 0, because the "PLA/PETG" (or soluble supports) mode is activated by this value (from my understanding). And it doesn't take unsupported surfaces into account.

Still present in 2.5.0-nightly (eb5b9a77):

[image]

Felix14-v2, 2026-10-06:
@ianalexis I think that may be relevant to the [smart material type usage](https://github.com/OrcaSlicer/OrcaSlicer/pull/14495) PR