OrcaSlicer/OrcaSlicer #3913: Crash when trying to reprint remotely.

nakwada, 2024-01-31:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Printer

Bambulab P1S with AMS

### How to reproduce

1. Go to Device > MicroSD Card 
2. Go to Model tab
3. Click Print on the thumbnail of any already printed plate.
4. Hanging for a few seconds then crash, OrcaSlicer closes

### Actual results

OrcaSlicer closes leaving some communication processes open with the printer.
When I retry the operation a few times, Orca crashes and at some point indicates the printer is already in too many conversations.
Still cannot restart a print at this point. Only solution is to reslice and resend to print.

### Expected results

It should not close ^^
Under normal conditions the window to select the spool in the AMS and confirm the print should appear.

### Project file & Debug log uploads

[2023.01.31 - OrcaSlicer crash logs.zip](https://github.com/SoftFever/OrcaSlicer/files/14117028/2023.01.31.-.OrcaSlicer.crash.logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

nakwada, 2024-02-01:
Additional informations about the printer:
Printer is up to date with the latest firmware as of February 2024.
Printer is connected to same network as laptop.

Jer406, 2024-03-12:
im having exactly the same issue

dartrax, 2024-05-01:
Also reported in (later) #4899.

closed 2024-08-09 by github-actions[bot]