OrcaSlicer/OrcaSlicer #3915: Seam Edge: Highlight Angle Option, Sharpest Corner

TheAtomicSoul, 2024-01-31:
Please add two options for Seams: 

1:  Seam Position: Add "Sharpest Corner"

2: Seam Painting: Add "Highlight Angle" option to allow automatic selection based on angles, similar to how  "Highlight Overhangs" works for support painting.

TheAtomicSoul, 2024-05-01:
Would really like this option added: I'd think it's implementation wouldn't be too difficult.

Vantage-DS, 2024-06-07:
Seconded, I have major issues with seam painting.
Cura's seam adaptability is much better.

Nandox7, 2024-06-14:
Plus one to this.
Getting problems printing a print in place part with gears, slicer decides to move the seam to where the gears mesh as it hides it. That is true but causes the gears to fuse together due to this.

[image]
[image]

Opposed to Creality Print slicer that has the option "Sharpest Corner" that moves it away from there and the part prints well.
[image]

closed 2024-09-20 by github-actions[bot]

hmenzagh, 2024-11-01:
This is the biggest flaw of orca imo, why are we so little asking for this ?

TheAtomicSoul, 2024-11-08:
Is there any way to reopen this?

Nandox7, 2024-11-08:
Open a new one and link to this one?
Makes no sense this was closed.

Nandox7, 2024-11-08:
Opened #7414