OrcaSlicer/OrcaSlicer #3922: More calibration and quality tests predefined

Sonnenwerk, 2024-02-01:
I would be exited, if we could have more calibration and quality test already prepared in the slicer.

I REALLY MISS THE BENCHY !

mrmultifunk, 2024-02-01:
The benchy is already there, right-click on the buildplate --> Add Handy models

Sonnenwerk, 2024-02-01:
Whooo great ! A hidden feature - nice !

Then i would just have the request to add them to the "calibration" menu too.

DVSVIDEO, 2024-02-03:
> Whooo great ! A hidden feature - nice !
> 
> Then i would just have the request to add them to the "calibration" menu too.

Printing Benchy is not really calibration...I don't think it belongs there.
Calibration is an interactive process: Print...setting....print...setting

closed 2024-05-11 by github-actions[bot]