OrcaSlicer/OrcaSlicer #3926: adding support of GLB/GLTF file format

Makrel, 2024-02-01:
I understand that this may not be the most important and necessary feature in OrcaSlicer, but in my modeling program it is most convenient to work with 3D models with the file format GLB/GLTF. I mainly use Сura for slicing, and it supports this format... I am slowly going to switch completely to OrcaSlicer, but support for this format is a bit lacking. Although I can work with STL, but it is more inconvenient.
So if there is an opportunity and it will not be too time-consuming, please consider it.

tajnymag, 2024-08-07:
> Orca bot: this issue is stale because it has been open for 90 days with no activity.

Bad bot

closed 2024-11-13 by github-actions[bot]

CjlaBuk, 2025-09-03:
Just do it!

john-clark, 2026-01-17:
With color!