OrcaSlicer/OrcaSlicer #3927: Stl vs Step files

olcayoren, 2024-02-01:
### OrcaSlicer Version

1.9.0

### OS version

Macos Sonoma 14.0

### Additional system information

Macbook arm m1 2020

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

Is it normal that there is such a difference between step and stl files?
[image]
[image]

### Actual results

None

### Expected results

None

### Project file & Debug log uploads

[CaseV2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14131084/CaseV2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Eldenroot, 2024-02-01:
yes, .stl is much simpler... thats why I prefer .step files (also for exporting models from Fusion etc).

Vorlent, 2024-09-16:
> > yes, .stl is much simpler... thats why I prefer .step files (also for exporting models from Fusion etc).
> 
> Bambu Studio / Orca are just taking your step file and converting it into a faceted / STL file, but with no control over the quality. I get FAR superior quality exporting STLs from my CAD program than with importing a STEP into BS/Orca.
> 
> I reported this here: [bambulab/BambuStudio#3437](https://github.com/bambulab/BambuStudio/issues/3437)

In this issue, the STL quality is objectively worse compared to STEP.

STL:
[image]

STEP:
[image]

The first picture has the same problems in the STL preview.

CCS86, 2024-09-16:
That has to do with your STL export settings.

GlauTechCo, 2024-09-18:
> That has to do with your STL export settings.

I use onshape as a drawing program. I choose the highest quality offered as stl output quality.
[image]

CCS86, 2024-09-18:
> I use onshape as a drawing program. I choose the highest quality offered as stl output quality. >

Onshape's STL translation is still causing that issue.

dewi-ny-je, 2024-10-21:
> > I use onshape as a drawing program. I choose the highest quality offered as stl output quality. >
> 
> Onshape's STL translation is still causing that issue.

It would be good if at least Orca provided some information about the meshing performed internally, so I can at least know which settings in my CAD software will result in a better model.

silvernull, 2024-11-21:
It would be useful if you could edit the quality settings,

the surface, normal deviation, max length of edge in mm, max aspect ratio of triangles (the controls fusion 360 gives)

movinimage, 2024-12-08:
How come we can’t access the meshing settings that BS offers on STEP import?
[image]

movinimage, 2024-12-08:
BS added this dialog in the 1.10 public beta on Oct 15.

silvernull, 2025-01-24:
> How come we can’t access the meshing settings that BS offers on STEP import?

Really hoping we get this in 2.3.0 so I never have to use 20+MB STL files ever again.

1951FDG, 2025-02-19:
Me as well!! Will this be included in 2.3.0?