OrcaSlicer/OrcaSlicer #3933: Trying to build on Gentoo...hit a wall: "‘wxWebRequest’ does not name a type"

salfter, 2024-02-02:
### OrcaSlicer Version

master

### OS version

Gentoo Linux

### Printer

haven't gotten that far

### How to reproduce

```
sudo emerge -1 media-libs/glfw dev-libs/cereal sci-libs/nlopt media-gfx/openvdb # there may be others; these are what I needed to add
sudo eselect wxwidgets set gtk3-unicode-3.2-gtk3 # emerge x11-libs/wxGTK:3.2-gtk3 if this fails
git clone https://github.com/SoftFever/OrcaSlicer/
cd OrcaSlicer
mkdir build
cd build
cmake -DOPENVDB_FIND_MODULE_PATH=/usr/lib64/cmake/OpenVDB -DSLIC3R_PCH=0 -DSLIC3R_WX_STABLE=1 -DSLIC3R_GTK=3 ..
make
```

### Actual results

compile completes without errors

### Expected results

```
In file included from /home/salfter/OrcaSlicer/src/slic3r/GUI/StatusPanel.hpp:7,
                 from /home/salfter/OrcaSlicer/src/slic3r/GUI/Monitor.hpp:46:
/home/salfter/OrcaSlicer/src/slic3r/GUI/SliceInfoPanel.hpp: At global scope:
/home/salfter/OrcaSlicer/src/slic3r/GUI/SliceInfoPanel.hpp:47:5: error: ‘wxWebRequest’ does not name a type
   47 |     wxWebRequest web_request;
      |     ^~~~~~~~~~~~
/home/salfter/OrcaSlicer/src/slic3r/GUI/SliceInfoPanel.hpp:84:30: error: ‘wxWebRequestEvent’ has not been declared
   84 |     void on_webrequest_state(wxWebRequestEvent &evt);
      |                              ^~~~~~~~~~~~~~~~~
```

(this error is repeated at line 492)

### Project file & Debug log uploads

No such files...can't produce them as it's not even building, let alone running.

There were also errors in src/slic3r/GUI/Printer/PrinterFileSystem.cpp where ```std::string```s were being fed to functions that expect ```const char*```s.  This patch fixes that, but now I've run up against a bug I can't yet figure out.

```
diff --git a/src/slic3r/GUI/Printer/PrinterFileSystem.cpp b/src/slic3r/GUI/Printer/PrinterFileSystem.cpp
index 886421c65..fbe31f172 100644
--- a/src/slic3r/GUI/Printer/PrinterFileSystem.cpp
+++ b/src/slic3r/GUI/Printer/PrinterFileSystem.cpp
@@ -633,7 +633,7 @@ void PrinterFileSystem::DownloadNextFile()
                 download->ofs.open(download->local_path, std::ios::binary);
                 if (!download->ofs) {
                     download->error = last_system_error();
-                    wxLogWarning("PrinterFileSystem::DownloadNextFile open error: %s\n", wxString::FromUTF8(download->error));
+                    wxLogWarning("PrinterFileSystem::DownloadNextFile open error: %s\n", wxString::FromUTF8(download->error.c_str()));
                     return FILE_OPEN_ERR;
                 }
             }
@@ -644,7 +644,7 @@ void PrinterFileSystem::DownloadNextFile()
             download->ofs.write((char const *) data, size);
             if (!download->ofs) {
                 download->error = last_system_error();
-                wxLogWarning("PrinterFileSystem::DownloadNextFile write error: %s\n", wxString::FromUTF8(download->error));
+                wxLogWarning("PrinterFileSystem::DownloadNextFile write error: %s\n", wxString::FromUTF8(download->error.c_str()));
                 return FILE_READ_WRITE_ERR;
             }
             download->boost_md5.process_bytes(data, size);
@@ -1117,7 +1117,7 @@ void PrinterFileSystem::RecvMessageThread()
             auto & msg = m_messages.front();
             // OutputDebugStringA(msg.c_str());
             // OutputDebugStringA("\n");
-            wxLogInfo("PrinterFileSystem::SendRequest >>>: \n%s\n", wxString::FromUTF8(msg));
+            wxLogInfo("PrinterFileSystem::SendRequest >>>: \n%s\n", wxString::FromUTF8(msg.c_str()));
             l.unlock();
             int n = Bambu_SendMessage(m_session.tunnel, CTRL_TYPE, msg.c_str(), msg.length());
             l.lock();
@@ -1158,7 +1158,7 @@ void PrinterFileSystem::HandleResponse(boost::unique_lock<boost::mutex> &l, Bamb
     json        root;
     //OutputDebugStringA(msg.c_str());
     //OutputDebugStringA("\n");
-    wxLogInfo("PrinterFileSystem::HandleResponse <<<: \n%s\n", wxString::FromUTF8(msg));
+    wxLogInfo("PrinterFileSystem::HandleResponse <<<: \n%s\n", wxString::FromUTF8(msg.c_str()));
     std::istringstream iss(msg);
     int                cmd    = 0;
     int                seq    = -1;
@@ -1256,10 +1256,10 @@ void PrinterFileSystem::Reconnect(boost::unique_lock<boost::mutex> &l, int resul
         std::string url = m_messages.front();
         m_messages.clear();
         if (url.size() < 2) {
-            wxLogMessage("PrinterFileSystem::Reconnect Initialize failed: %s", wxString::FromUTF8(url));
+            wxLogMessage("PrinterFileSystem::Reconnect Initialize failed: %s", wxString::FromUTF8(url.c_str()));
             m_last_error = atoi(url.c_str());
         } else {
-            wxLogMessage("PrinterFileSystem::Reconnect Initialized: %s", Slic3r::GUI::hide_passwd(wxString::FromUTF8(url), {"authkey=", "passwd="}));
+            wxLogMessage("PrinterFileSystem::Reconnect Initialized: %s", Slic3r::GUI::hide_passwd(wxString::FromUTF8(url.c_str()), {"authkey=", "passwd="}));
             l.unlock();
             m_status = Status::Connecting;
             wxLogMessage("PrinterFileSystem::Reconnect Connecting");
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

cgf1, 2024-04-29:
Just got the same error.  I tried this:
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -123,6 +123,8 @@ if (MSVC)
     add_compile_options(/wd4244 /wd4267)
 endif ()
 
+add_compile_options(-DwxUSE_WEBREQUEST)
+
 if (${CMAKE_CXX_COMPILER_ID} STREQUAL "AppleClang" AND ${CMAKE_CXX_COMPILER_VERSION} VERSION_GREATER 15)
     add_compile_definitions(BOOST_NO_CXX98_FUNCTION_BASE _HAS_AUTO_PTR_ETC=0)
 endif()

fkcroot, 2024-05-02:
Thats a Gentoo side misconfiguration, wxGTK should be built against `curl` use flag to enable webrequest support.

Confirmed on x11-libs/wxGTK/wxGTK-3.2.2.1-r3 and 400220d7 OrcaSlicer rev.

closed 2024-08-09 by github-actions[bot]