OrcaSlicer/OrcaSlicer #3939: Find an replace option

Galloant, 2024-02-02:
**Problem**
I have changed from Prusa slicer to Orca just recently. 2 of my 3 printers work fine with it, but when importing the gcode output from Orca into the Anker slicer, it doesn't recognize the print movements type (perimeter, infill, etc.).
I found this to be caused by the different type of naming scheme such as  "Outer wall" vs. "External perimeter" for example.
This results in Anker slicer not being able to generate a AI file and thus the AI recognition doesn't work.
[image]

(This is my best guess until now, so please feel free to correct me on that.)

**Solution**
Replace the type names when creating the gcode. This would be easiest if there would be a option to search and replace text, like in Prusa Slicer:
[image]

Galloant, 2024-02-02:
Circumvented the issue by using AnkerMake Studio instead. AnkerMake Slicer is obsolete as it seems.
Nevertheless, option to find and replace text would still be appreciated.

atmacali, 2024-04-08:
same problem i need find and replace to set fan number as P2. I need to control cooling fan as M106 P2 S255

closed 2024-07-15 by github-actions[bot]