OrcaSlicer/OrcaSlicer #3940: Being able to set fan speed threshold without needing to change minimum layer time

felipejfc, 2024-02-02:
**Is your feature request related to a problem? Please describe.**
I wanted to be able to do the following:
-> If layer time is less than 8 seconds, use fans at 100%
-> Set minimum layer time to 3 seconds

Today I can only change both things at once, if minimum layer time is 3 seconds, this will be the threshold to fans at 100%.

**Which printers will be beneficial to this feature**
All printers. I'm using a Ender 3 V3 SE with Klipper

**Describe the solution you'd like**
In creality print they have both options unlinked, minimum layer time is a separate option than max fan layer time threshold. We could do the same.

**Describe alternatives you've considered**
I don't think there's an alternative today in orca, but happy to hear from others if I'm wrong.

DVSVIDEO, 2024-02-03:
At any time you can specify the fan speed for the layer time.
You can slow down printing for better cooling.
If these settings are not sufficient, you should think about hardware modification of the cooling...

[image]

felipejfc, 2024-02-03:
Makes sense but I wanted to be able to change minimum layer time without changing max fan speed threshold. Isn't this reasonable?

closed 2024-05-11 by github-actions[bot]

patrezp, 2024-12-04:
I just saw induction hotend came to reality, and having temperature modifier would be great addition to printing 0 gap supports for example. See https://www.youtube.com/watch?v=XphpaHd8Q9s for more ideas