OrcaSlicer/OrcaSlicer #3942: Allow for different global settings and printer assignment per plate

thehunterofshadows, 2024-02-02:
**Is your feature request related to a problem? Please describe.**
The multi-plate feature is great, but hampered by the lack of customization per plate.  Example: I want to print a mini-billards table.  Want to do the table, and legs on a QIDI tech with 0.2mm layer height, do the pool cues and balls on an Ender 3 with 0.1mm layer height.  This would have to be two separate projects.

**Which printers will be beneficial to this feature**
All printers - when there is a need to setup different global settings per plate, and use different bed sizes.

**Describe the solution you'd like**
Split global settings, and printer selection by plate.  Allowing the use of different bed sizes, and settings like layer height per plate.

**Describe alternatives you've considered**
Without this feature, separate projects would have to be used for each global setting change or each printer change, reducing the value and the clean approach of a multi-plate project.

**Additional context**
I will gladly answer any questions.

giveen, 2024-02-02:
I requested the same thing 

https://github.com/SoftFever/OrcaSlicer/issues/3119

thehunterofshadows, 2024-02-03:
> I requested the same thing
> 
> #3119

I think that's a little different.  I'm not suggest multiple profiles on same plate, i'm suggesting different printers and different settings plate to plate.  Like plate one is 0.2 layer height and using QIDI tech printer with a large bed, and plate two is 0.1 layer height and it using an ender 3 with a smaller bed

giveen, 2024-02-03:
Ah. In some ways similar.

closed 2024-05-11 by github-actions[bot]