OrcaSlicer/OrcaSlicer #3945: Сreality-SR10-SE printer profile

teleskop-tv, 2024-02-02:
Please add a Сreality-SR10-SE printer profile into your wonderful slicer program! Please! A lot of people already have this printer. We have been waiting for a profile for this printer for a very long time. Please add a Creality-SR10-SE printer profile. Please, please, please!!!

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

V1os, 2024-03-06:
Yes, or sey how preset CR-10 SE? From witch printer base?

gamali15, 2024-03-13:
Yes, we have been waiting for a long time for the profile to be loaded. we cannot use the printer with full efficiency.

henrivdr, 2024-03-19:
In the meantime you can look inside the Benchy gcode, which shipped with the memory stick, for the printer settings that was used.

cochcoder, 2024-03-28:
Is Сreality-SR10-SE a mistype, do you mean Сreality-CR10-SE?

teleskop-tv, 2024-03-29:
Yes, of course! I mean Reality-CR10-SE. I apologize for the mistake I made in the 3d printer model. Please add a Сreality-СR10-SE printer profile into your wonderful slicer program!

cochcoder, 2024-03-29:
> Yes, of course! I mean Сreality-CR10-SE. I apologize for the mistake I made in the 3d printer model. 

Don't worry about it, just wanted to make sure that I wasn't confusing two completely different 3D printers!

> Please add a Сreality-СR10-SE printer profile into your wonderful slicer program!

I have a PR request for a standard profile, if you want to give settings for a draft or detailed one, please do!

teleskop-tv, 2024-03-29:
Thank you! I'm not sure I can figure out what I need to do. If it is not difficult for you, please add the profile of this printer to the distribution package of the program. Night assembly or something like that

PR #4757 (2024-03-29): Add Creality CR-10 SE
Closes  #4746, closes  #3945

The base .json file and bed model/texture, as well as the cover, is based on #3749, while the standard print settings are based on @Adilrafiq001's [print profile](https://www.printables.com/model/709820-creality-cr10-se-orca-slicer-profile/files) and speed, acceleration, and some other printer settings, like retraction, are based off [phzwart's settings](https://www.thingiverse.com/thing:6349830/files).
closed 2024-03-29 by SoftFever