OrcaSlicer/OrcaSlicer #3948: Auto Orient is not working as it used to

Darthmonkey, 2024-02-03:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1. Import a STEP file, note that the orientation is not correctly optimised for the model.
2. Click 'Auto Orient', note the model doesn't change.
3. Select 'Lay On Face' and select a suitable face for it to lay on, then click 'Auto Orient' and note it doesn't change.
4. Select an unsuitable face for it to lay on, click 'Auto Orient' and note it usually selects a suitable face.

### Actual results

Reproduction description details actual results for convenience.

### Expected results

After importing a model which is not optimally orientated, 'Auto Orient' should choose the 'best' orientation to avoid supports as much as possible.

### Project file & Debug log uploads

No project or debug files are relevant in this case.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ohmarinus, 2024-02-04:
I'm surprised there hasn't been a report on this before. I can confirm, for me the `Auto Orient` function hasn't worked in the last full release. No matter what kind of object I add. I always have to do it manually by selecting a face.

kylek29, 2024-02-04:
^^ It's been reported, and I think I saw it pass through the PR, so it might be fixed already. There was also a sister issue where a Lay-on-Face selection would sometimes rotate the model 22.5°.

pbouna321, 2024-02-06:
> I'm surprised there hasn't been a report on this before. I can confirm, for me the `Auto Orient` function hasn't worked in the last full release. No matter what kind of object I add. I always have to do it manually by selecting a face.

Having the exact same experience.

closed 2024-05-15 by github-actions[bot]