OrcaSlicer/OrcaSlicer #3949: Flatpak Application

powpingdone, 2024-02-03:
I am requesting permission to create a Flatpak application ([manifest here](https://github.com/powpingdone/flathub/tree/orcaslicer)), along with any modifications required by other contributors. Given permission, I will go on to flathub and publish the application.

powpingdone, 2024-02-05:
@SoftFever I'm very sorry for pinging you, but I would like your "ok" on this.

SoftFever, 2024-02-05:
Hi @powpingdone 

First thing first, that's great!

Could you please clarify what permission you are asking for? At the moment, Orca's CICD only creates appimages.
If you are working on building or GitHub actions scripts, would you mind submitting a pull request? Please let me know if you meant something else.

powpingdone, 2024-02-05:
From https://docs.flathub.org/docs/for-app-authors/submission#theres-an-app-that-id-like-to-see-on-flathub-but-im-not-the-developer

> *If there’s an app that you'd like to be distributed on Flathub, the best first course of action is to approach the app’s developers and ask them to submit it.* Remember that for some projects or communities, this may be the first that they have heard of the Flatpak technology or interacted with somebody who advocates it. When you are making such requests, please be aware that the people you are talking to may be forming their first impressions of Flatpak as a technology and a community. 

I could do initial distribution of both nightly, and (eventually) stable releases past v2 (due to that flatpak patch needed) using CI, but since I'm not the owner of the repository I would like your permission to operate this.

powpingdone, 2024-02-05:
I'm not asking you to submit a flatpak, I would be doing that myself (with input from you and other devs).

SoftFever, 2024-02-05:
> From https://docs.flathub.org/docs/for-app-authors/submission#theres-an-app-that-id-like-to-see-on-flathub-but-im-not-the-developer
> 
> > **If there’s an app that you'd like to be distributed on Flathub, the best first course of action is to approach the app’s developers and ask them to submit it.** Remember that for some projects or communities, this may be the first that they have heard of the Flatpak technology or interacted with somebody who advocates it. When you are making such requests, please be aware that the people you are talking to may be forming their first impressions of Flatpak as a technology and a community.
> 
> I could do initial distribution of both nightly, and (eventually) stable releases past v2 (due to that flatpak patch needed) without using CI, but since I'm not the owner of the repository I would like your permission to operate this.

Thanks for the explanation.

Currently, all OrcaSlicer builds, including the official releases, are fully automated within this GitHub repository and it's CI/CD pipeline to ensure security. 
I intend to maintain this approach for any new distribution formats, purely from a security and maintenance perspective. 
I hope you can understand my rationale.

powpingdone, 2024-02-05:
That's completely understandable. Shall I setup/pr a CI job to distribute a flatpak then?

SoftFever, 2024-02-06:
> That's completely understandable. Shall I setup/pr a CI job to distribute a flatpak then? I would have to give you a key as a secret to allow uploading to flathub.

👍 
That would be great!

ClaytonRehder, 2024-02-07:
+1. Good initiative.

tapir, 2024-02-11:
this is great! can we not wait for 2.0 and just have one for 1.9 already?

powpingdone, 2024-02-11:
@tapir afaict, because of the flatpak specific patches, no.

hadess, 2024-02-26:
> I am requesting permission to create a Flatpak application ([manifest here](https://github.com/powpingdone/flathub/tree/orcaslicer)), along with any modifications required by other contributors. Given permission, I will go on to flathub and publish the application.

That manifest looks eerily familiar ;)

Let me know if you think there's anything we could do to make it easier to share between the 2 manifests repositories, once this has landed, and please CC: me on the Flathub review as well.

powpingdone, 2024-03-10:
shhhh I totally didn't do anything to it :P

kugelzucker, 2024-03-30:
so as it happens i just opened the orcaslicer github page to check if there was a release for 2.0 since bambustudio messed up some tree supports. 6 minutes ago was the release of 2.0, i am in luck.

since this is an open source project and devs are always very interested in annoying and pushy comments i thought i would do my part and ask:

why is there no flatpak released 6 minutes after the official release? i cant believe it.

seriously though, what is the process and how can we help? is this already implemented and the pipeline is just slower (6 minutes, i mean cmon ... :D)?

thanks for all your great work!

powpingdone, 2024-03-30:
@kugelzucker currently i'm bogged down by irl stuff and ci is failing for reasons unknown (when it was previously fine).

kugelzucker, 2024-05-07:
im giving this a nudge, checking in

kugelzucker, 2024-05-13:
> im giving this a nudge, checking in

also, this builds and installs fine for me. fyi

powpingdone, 2024-05-13:
@kugelzucker tl;dr: this messes with a lot of dep resolution stuff so it gets messy on other systems. it's fine for flatpak, but it's not fine for literally everyone else

powpingdone, 2024-05-25:
Update for everyone who is following this but not the PR: #4175 has been merged. Next up, Softfever and I will have to get in contact with Flathub to get the necessary things (api keys) to upload from CI.

powpingdone, 2024-05-25:
cc #5449

powpingdone, 2024-05-29:
cc #5474

cochcoder, 2024-05-29:
I'd also like to note that you may want to build the flatpak with freedesktop-22.08 rather than gnome-45, as freedesktop prioritizes having applications work consistently on all platforms rather than one. I also noticed in my testing that building with freedesktop causes the flatpak to have a slightly smaller size and uses less ram than the gnome build.

powpingdone, 2024-06-16:
@hgelpke cross quoting from pr:
> Can somebody give a rundown on how to build this? Tried on Fedora 40 and failed but it is also the first time I'm trying this.
tldr: `flatpak-builder --force-clean --user --install --jobs=2 builddir io.github.softfever.OrcaSlicer.yml`. 
Adjust `--jobs=` to fit in RAM, as a single compile job in OrcaSlicer can take up 4GB ram or more.
Build is currently broken and i'm investigating in #5510.

hgelpke, 2024-07-09:
> @hgelpke cross quoting from pr:
>
>
> tldr: `flatpak-builder --force-clean --user --install --jobs=2 builddir io.github.softfever.OrcaSlicer.yml`. Adjust `--jobs=` to fit in RAM, as a single compile job [...]

Great! Thanks for the reply and apologies for missing it. Good luck on the fix and I appreciate your efforts

powpingdone, 2024-07-18:
Ok, after the cmake fiasco we're now in the tree. Finally. Going to file an issue to flathub soon(tm).

albei, 2024-07-18:
P1S camera still not working Flatpak 2.1.1 Build all #2600
Testing it on a fedora silverblue.
"Player is malfunctioning. Please reinstall the system player. [2:2]"
Even so we finally have a flatpak! Many thanks for your effort !

powpingdone, 2024-07-18:
@albei tracked in a separate issue #5449, iirc that's a external patch currently. I do want to do the gstreamer stuff properly.

powpingdone, 2024-07-25:
https://discourse.flathub.org/t/orcaslicer-requesting-external-upload-to-flathub/7340 @SoftFever (also @hadess because I don't know where to go with this situation).

tapir, 2024-08-11:
Does this have to be fully automated through CI @SoftFever? What if flathub doesn't want to do an external upload?

JustinLex, 2024-08-17:
Is there an issue with letting Flathub build the flatpak themselves? It's important for security reasons that Flathub is able to build from source, and the build process and configuration is identical to what is already in the Github Actions CI. Flatpak builds can continue to run in Github Actions CI as an internal testing step, but it looks like the build distributed by Flathub should also be built by Flathub.

tapir, 2024-08-25:
Looks like the issue on flathub side was closed due to inactivity. I'd like to take this over if possible but I, 100%, agree with @JustinLex, I don't see any reason to not follow flathub standards. Are we missing something @powpingdone?

powpingdone, 2024-08-25:
@tapir no it's just been me being incredibly busy. I'll make the flathub issue proper (not discourse).

powpingdone, 2024-08-25:
I misread your comment, sorry. @SoftFever why can't we have flathub building the flatpak for orcaslicer externally?

powpingdone, 2024-08-28:
This would make distribution a lot easier because you can just submodule the repo to them and still have CI here.

powpingdone, 2024-09-09:
@SoftFever bump, waiting for your input.

Ziomal12, 2024-10-08:
Has anyone got working flatpak of 2.2?

cochcoder, 2024-10-23:
@powpingdone 

I don't mean to be that person, but do you know when we could get a working GitHub action for the flatpak?

anarsoul, 2024-10-24:
@SoftFever sorry for nagging you, but could we please get an approval for building flatpak externally?

It would simplify installation of Orca Slicer for a lot of Linux users and personally I don't see any drawbacks for you, only benefits.

Flatpak runtime environment is more contained than any Linux distro, so Orca Slicer will be running more stable and it would be easier to reproduce bugs if there are any.

Ziomal12, 2024-10-25:
@anarsoul Correct me if I'm wrong but as far as I understand under AGPL-3.0 licence distribution of the software is allowed as you disclose source and use the same licence as original.

anarsoul, 2024-10-25:
@Ziomal12 see flathub submission guide: https://docs.flathub.org/docs/for-app-authors/submission/#theres-an-app-that-id-like-to-see-on-flathub-but-im-not-the-developer

We are at the step one. Submitting it without author's approval is the last resort.

cochcoder, 2024-10-25:
@ChaosBlades 
It's only been 12 hours, give SoftFever some time. Both SoftFever and PowPingDone are really busy.

tapir, 2024-10-25:
@cochcoder we haven't heard from him on this thread since Aug. That's probably what @ChaosBlades meant.

anarsoul, 2024-10-27:
I created a repo with (fixed) flatpak manifest for OrcaSlicer: https://github.com/anarsoul/io.github.softfever.OrcaSlicer

Flatpak for v2.2.0-rc can be found here: https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-rc

cochcoder, 2024-10-27:
@anarsoul 

Do you think you'll be able to make a PR with the manifest fixes?

anarsoul, 2024-10-27:
@cochcoder I can, although it won't help with getting flatpak into flathub.

One of the fixes has been sitting in review for 3 weeks: https://github.com/SoftFever/OrcaSlicer/pull/7057

The other fix is getting rid of python script and integrating dark theme detection into wxWidgets (see patch in my repo, I took it from bambulab slicer manifest repo)

cochcoder, 2024-10-27:
> @cochcoder I can, although it won't help with getting flatpak into flathub.
> 

That's fine, we would've need to get a working flatpak first anyways. 

> 
> One of the fixes has been sitting in review for 3 weeks: https://github.com/SoftFever/OrcaSlicer/pull/7057
> 
> 
> 
> The other fix is getting rid of python script and integrating dark theme detection into wxWidgets (see the patch in my repo, I took it from bambulab slicer manifest repo)
> 
> 

You could link to those in your PR and mention that they need to get merged alongside the manifest fixes.

anarsoul, 2024-10-28:
> If the OrcaSlicer devs are anything like me, they're probably expecting better commit messages.

It's not my PR, but if OrcaSlicer devs were anything like me, they'd probably fix up commit message if they don't like it and merge the PR :)

trebory6, 2024-11-18:
> @ChaosBlades It's only been 12 hours, give SoftFever some time. Both SoftFever and PowPingDone are really busy.

It's been 3 weeks, is that enough time? haha

Seriously though, it's disappointing that flatpak compatibility is taking so long. I'm getting very tired of Orcaslicer breaking with each new Kernel update.

anarsoul, 2024-11-18:
> Seriously though, it's disappointing that flatpak compatibility is taking so long officially. I'm getting very tired of bandaging Orcaslicer's appimage with environment variables after it breaks with each new Kernel update.

Just install flatpak from https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-1

anarsoul, 2024-11-29:
@SoftFever sorry to nag you again, but can we get this issue moving? If you need to discuss it further in real-time, text or voice, I'm available in discord on Orca Slicer server, same nickname as here.

Skaifer, 2024-12-04:
Feels so right to install a flatpak. I have issue though with flatpak version as well. The preview side is all gray. Terminal output - https://paste.pika-os.com/p/sloth-camel-ape. Tried to set env variables as for appimage, but that doesn't do anything or makes things worse. And appimage currently runs with those fixes, but preview is terribly slow.

Jookia, 2024-12-04:
As a user I would really appreciate an official Flathub package even if it's not created by the OrcaSlicer devs.

Long term this might be the more secure option too: Right now to run OrcaSlicer I have to download binaries with no way or ability to check if they're built from corresponding source code. I can't even check if the binaries are the ones the developer uploaded, there's no package signing.

Having an independent third party such as Flathub build and distribute binaries based on a manifest would give me an ability to trust the version of OrcaSlicer I'm running.

Skaifer, 2024-12-20:
Ran Flatpak version successfully on Bazzite under Wayland, thanks to [Fr0sty4 comment](https://github.com/SoftFever/OrcaSlicer/issues/6900#issuecomment-2480244457):
> For anyone using the flatpak -- adding this environment variable via Flatseal should work for software rendering and stays inside the flatpak:

`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json`

The thing, that's different from previous variable, is the path to 50_mesa.json, which previously was `/usr/share/glvnd/egl_vendor.d/50_mesa.json`. I checked the new path, and there is nothing. Basically giving it nothing fixed the issue, which is weird, but I digress. The viewport when viewing the sliced part is slow, but that's what we get until it gets fixed for Nvidia users.

anarsoul, 2024-12-21:
> until it gets fixed for Nvidia users

Someone needs to report it to NVIDIA, it's very likely to be their bug

Skaifer, 2024-12-21:
[MackBambu](https://github.com/bambulab/BambuStudio/issues/4669#issuecomment-2495929679) at BambuStudio thread says Nvidia has dropped old OpenGL version support, so they are now working on a fix, which hopefully would eventually come to OrcaSlicer and heal our sores.

egruttum, 2025-01-11:
> I created a repo with (fixed) flatpak manifest for OrcaSlicer: https://github.com/anarsoul/io.github.softfever.OrcaSlicer
> 
> Flatpak for v2.2.0 can be found here: https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-1

Thank you for this.  I drove myself crazy trying to make OrcaSlicer work on my Chromebook with the AppImage.  This worked so easily.  Hopefully this gets into the official Flatpak repository someday to make it even easier for people to find.

Prazgreen1, 2025-03-02:
> Thank you for this. I drove myself crazy trying to make OrcaSlicer work on my Chromebook with the AppImage. This worked so easily. Hopefully this gets into the official Flatpak repository someday to [...]

this is in the official repo

mtorromeo, 2025-03-02:
> this is in the official repo

Which official repo are you talking about?

It's not here https://flathub.org/apps/search?q=orcaslicer

616b2f, 2025-03-02:
For those who are not aware there is already a Flatpak file in the release bundle it's easy to install, just download and open it with you Software Store application (it's discovered by the most distros automatically, when you try to open an XY.flatpak file.

I hope this helps someone, as I was also searching for a way to use the Flatpak version of Orca.

CharlemagneLasse, 2025-03-02:
> For those who are not aware there is already a Flatpak file in the release bundle it's easy to install, just download and open it with you Software Store application (it's discovered by the most [...]

Such a flatpak is only really useful when it is auto-updating and can actually be found by the community using the builtin flatpak repo management tools.

And https://github.com/anarsoul/io.github.softfever.OrcaSlicer/ seems to be the only one which seems to work. It might therefore be good to use this one on flathub.

The attempt to do this was unfortunately blocked by @SoftFever (for unknown reasons because no feedback is given by @SoftFever)

Jookia, 2025-03-02:
I believe the reasoning is that SoftFever doesn't want people to package it externally, so someone would have to make a fork and rebrand it for Flathub.

[36 more replies]