OrcaSlicer/OrcaSlicer #3952: Brim/skirt startpoint the same placement as rest of model

FlowForm, 2024-02-03:
The starting point of the brim and skirt is always in the left front corner and not near the rest of the chosen or painted seam position. 

This creates unnecessary and unwanted travel moves in the first layer. With filament that sticks really badly, the least amount of travel is preferred. Also with woodfil and other filaments this travel leaves ooze on the buildplate. 

[image]

- Wanted: Same startpoint of brim & skirt as rest of model!

FlowForm, 2024-05-06:
Would be so nice to see this implemented :-)

DorjeDorf, 2024-07-06:
bumping this in hopes of getting this sorted out / explained / implemented :D

Dumplap, 2024-08-10:
bumping

closed 2024-11-16 by github-actions[bot]