OrcaSlicer/OrcaSlicer #3955: "Collapse Sidebar"

ThinkBigPrintHuge, 2024-02-03:
**Is your feature request related to a problem? Please describe.**

Orca v1.9.0, I hate the new collapsing sidebar. I would love to see the default placement of the sidebar in the same location as previous versions of Oca. As of now when I open Orca, the side bar is halfway collapsed and it's not a smooth experience sliding it. Also, after clicking <<>> to collapse and then again to expand, it doesn't revert back to the position it was collapsed from. All very minor QOL things, I know. Thank you!

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

discip, 2024-02-03:
Have you tried the latest build: https://github.com/SoftFever/OrcaSlicer/actions/runs/7725298637

ThinkBigPrintHuge, 2024-03-02:
I finally installed 1.9.1. Moved the sidebar to see how docking (insert joke here) works. Now I have a free floating sidebar again and cannot figure out how to re-dock it.

closed 2024-06-09 by github-actions[bot]