OrcaSlicer/OrcaSlicer #3959: Please add CR-10S Pro V2 to the printers list

basem2b2, 2024-02-03:
**Is your feature request related to a problem? Please describe.**
It is very hard to add a new printer and figuring out the suitable settings and speed limits to that printer
**Which printers will be beneficial to this feature**
Creality CR-10S Pro V2
Please consider adding my printer in future updates.

Gurba667, 2024-03-08:
+1 on this.

Chaingum, 2024-03-15:
+1 it would be great to be able to use Orca with the cr-10s pro V1/V2 (V1 and V2 are quite the same)

cochcoder, 2024-06-14:
I'm starting to add support for this printer and was wondering if there were any changes that should be made from the PrusaSlicer profile, that is universally agreed upon.

cochcoder, 2024-06-16:
I have a build going [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/9538486374) with the 0.4 nozzle of the CR-10S Pro V2 added, please let me know if you see any issues or settings that should be added or changed for the global default. I plan on adding V1 of the CR-10S Pro and 0.6 nozzle size for both. 

**Make sure to have backups of your custom configs before trying this build, in case anything happens. If you're on windows, I recommend that you use the portable version rather than the installer for testing purposes**

BryanCAlcorn, 2024-08-14:
Will this get merged soon?

rwmccoy, 2024-11-13:
Has any additional work been done to finish this profile?

rwmccoy, 2025-02-12:
Ok, since no additional work seems to have been done to complete the profile, is there an existing profile that will work with the printer? Thank you.

geransmith, 2025-02-24:
I used the Pruse slicer profile and modified it for the Orca variables.

I've successfully had a few prints. I think it needs some tweaking around initial temps and such, but it is working at least. Mainly it was using the existing CR-10 V2 profile in Orca, then editing the Gcode.

Start G-code:
```
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S150 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling
M140 S[bed_temperature_initial_layer_single] ; set bed temp
M190 S[bed_temperature_initial_layer_single] ; wait for bed temp
G28 ; home all
BED_MESH_CALIBRATE PROFILE=default ; run a mesh calibration in Klipper
G1 Z2 F240
G1 X2 Y10 F3000
M104 S{first_layer_temperature[0]} ; set final nozzle temp
M109 S[nozzle_temperature_initial_layer] ; wait for extruder temp
G1 Z0.28 F240
G92 E0
G1 X2.0 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 F5000
G92 E0
G1 X2.3 Y10 E10 F1200 ; prime the nozzle
G92 E0
```

End G-Code:
```
{if max_layer_z < printable_height}G1 Z{z_offset+min(max_layer_z+2, printable_height)} F600{endif} ; Move print head up
G1 X5 Y{print_bed_max[1]*0.85} F{travel_speed*60} ; present print
{if max_layer_z < printable_height-10}G1 Z{z_offset+min(max_layer_z+70, printable_height-10)} F600{endif} ; Move print head further up
{if max_layer_z < printable_height*0.6}G1 Z{printable_height*0.6} F600{endif} ; Move print head further up
M140 S0 ; turn off heatbed
M104 S0 ; turn off temperature
M107 ; turn off fan
M84 X Y E ; disable motors
```

rwmccoy, 2025-02-24:
Thank you! This is a great starting point. 

I would still love to see it developed into an official profile.

closed 2025-06-02 by github-actions[bot]