OrcaSlicer/OrcaSlicer #3965: Pro Level Filament Features

kentonbmax, 2024-02-04:
**Is your feature request related to a problem? Please describe.**

- Every filament that is not using rfid, needs to be calibrated. Mainly flow dynamics and flow control. Run both at the same time on the same plate using the value from the first for the second. 
- Additionally the filament management is clunky. Each filament should be calibrated before use. Therefore make it easy to calibrate and store those settings. Suggested flow, Create new Filament --> Select it for calibration --> calibrate to it (if successful just freakin save the damn thing). 
- Make the filter fan have a delay before resuming normal operation after printing. Currently this does not work at all on Carbon X1. 
- Make heating the chamber work or provide a spot for custom gcodes for this one. Currently it does not work on Carbon X1. 

**Which printers will be beneficial to this feature**
All, CArbon X1

**Describe the solution you'd like**
1. Run both calibrations at the same time on the same plate using the value from the first for the second. 
- Additionally the filament management is clunky. Each filament should be calibrated before use. Therefore make it easy to calibrate and store those settings. Suggested flow, Create new Filament --> Select it for calibration --> calibrate to it (if successful just freakin save the damn thing). 
- Make the filter fan have a delay before resuming normal operation after printing. Currently this does not work at all on Carbon X1. 
- Make heating the chamber work or provide a spot for custom gcodes for this one. Currently it does not work on Carbon X1. 

Doing this will make orca even better regardless of which printer your using and save makers time!.

kylek29, 2024-02-04:
Re: 1) For what it's worth, this is the work flow for the X1C as they designed those tests to not interfere with eachother, it just has the extra step of saving the first tests value and then selecting it on the second step (intermediary). So that is something that could be further automated.

Re: Chamber Heating: I don't believe the X1C has a chamber heater. If has the chamber temperature, but there's no heater itself (afaik). This is why it uses the bed to heat the chamber for the Filament Dryer procedure.

kentonbmax, 2024-02-05:
Regarding the chamber heater, provide options to wait until desired temp is reached, and perhaps selecting heating mechanisms should a printer not have anything directly included for it. Putting printers inside of enclosures is fairly common these days for engineering materials.

kentonbmax, 2024-03-10:
Hmmm. It appears as though the FC has to be loaded after power cycling the printer. Possible to save in orca and associate with custom filament setting? Custom meaning generic that is calibrated and saved.

kentonbmax, 2024-03-10:
Another idea, allow users to set a global, chamber and custom start gcode for temp. Best to include a timer in this one for safety.

closed 2024-06-17 by github-actions[bot]