OrcaSlicer/OrcaSlicer #3966: Wrong STEP slicing when the same part STL is OK

sebox1, 2024-02-04:
### OrcaSlicer Version

1.6-1.8.1

### OS version

Windows 10

### Printer

Creality K1

### How to reproduce

Save the same solidworks model to STEP and STL

### Actual results

[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

it need to be fixed

### Project file & Debug log uploads

[STEP issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14157893/STEP.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

sebox1, 2024-02-04:
[image]

foreachthing, 2024-02-05:
There is no STEP (nor STL, nor SolidWorks) file in your ZIP. Could you please add those?

sebox1, 2024-02-05:
[ ! Orca STEP issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14167635/Orca.STEP.issue.zip)

sebox1, 2024-02-05:
Thanks for your interest. This trick didn't work for me when I saved STEP from my SW 2021sp5.1 again. Then, out of curiosity, I saved this STEP from SW 2019sp5.1 and now everything was OK.
Surprisingly, it means the SW 2021sp5.1 is to blame!

However, this discovery does not solve my problem in any way. It’s much easier for me to save this part as STL, while all the other 15 parts of the project are transferred to Orca as STEP and work without any problems.

Perhaps a little later I will install SW2023 or 24 and see how the STEP of this part, saved from a newer version, behaves.

But your observation became important and interesting. Thank you again!

sebox1, 2024-02-05:
I found the real cause of the bug!

In STEP Options of my SW 2021, two checkboxes are disabled, which should be enabled by default.

When I turned them back ON and saved the new STEP, Orca slice it without problems.

[image]

foreachthing, 2024-02-06:
You could write a small macro (vba, vb.net, c#) to make sure your settings are always the way you want/need them (lots of help and examples in the "API help" from SolidWorks).
I did not have the option "Split periodic faces" selected. Maybe that one is not needed? I don't even know what it does 😆 
Oh, and if you think this issue has been resolved, please close it 😉 .

closed 2024-02-06 by sebox1