OrcaSlicer/OrcaSlicer #3969: Automatic Speed Options...choose to use exact value or percentages

soundtechscott, 2024-02-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

If I want to change a speed for my material, I have to change every single line speed manually
but I already know there are simple math values that work for the other parts of my print
lets say as an example...
if I want to hit a max speed of 100mm/sec
I know I can proably go 50-60mm/sec on my first layer with no issues
my outer walls might only be 80mm...my inner walls might be the full 100mm
my infill will probably be 70-75mm

if my max speed is only 80mm/sec
then my first layer is going to be 40-48mm (50%-60%)
my outer walls are only going to be 64mm (80%)
my inner walls are going to be the full 80mm
my infill will be 56-60mm (70-75%)

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

all 3d printers

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Just make speed cells value or % like you have already done for line width

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I currently use an excel sheet to do my calc based on a percentage

**Additional context**
Add any other context or screenshots about the feature request here.

Tsoukan, 2024-02-07:
I was looking for this exact feature.  Being able to have the speed and acceleration values be calculated as a % of a base value would be fantastic  - also a document could be made to help new makers fill in starting % values.

soundtechscott, 2024-05-10:
lets keep this going as it hasnt been added yet and its a big quality of life feature for people that understand how it works

antonyburden, 2024-06-12:
100% is the setting I miss most from other slicers. Please consider adding this. Just 1 number to adjust will keep my profiles down to just 1 or 2.

closed 2024-09-18 by github-actions[bot]

Lbredeson, 2025-07-30:
Just seeing if this perma closed. Or if anyone has thought to bring this back as an issue. Would help with generating a constant flow rate.