OrcaSlicer/OrcaSlicer #3970: Multi Infill Direction

soundtechscott, 2024-02-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

would like to be able to change infill directions... maybe use a string [0,30,60,90,120,150]
and then a window to pick the number of layers at each direction before changing...

example would be do angle 0 for 10 layers, then do angle 30 for 10 layers...etc

this would help with strength of a lot of pieces

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

all printers would benefit

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

allow a string surrounded by [brackets]
give a seconday box for how many layers between each value

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I already do this in cura

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

mtc1122, 2024-05-01:
Maybe I'm misunderstanding your issue, but you can use the height range modifiers to change infill directions every certain distance. Strength>Advanced>Infill direction. Each progressive layer should be 90 degrees to the last, but you could have it change 30 degrees rotation every millimeter, for instance.

soundtechscott, 2024-05-01:
That function does not do what you describe
that function in Strength>Advanced>Infill direction
simply points to the direction you want the infill to follow, it doesnt rotate, it doesnt add to it....1 direction.....all lines that one direction, if I do 12* as an example, it doesnt do 12, 24, 36.... it just does 12, 12, 12, 12, 12, 12, 12, 12........
if I do a grid type pattern, it does 12, 102, 12, 102 (90* between each layer)

what I am looking for is for it to do like this
in this example set of pictures.... move by 30 each time
30,60,90, etc

[image]

[image]

[image]

[image]

bytemage, 2024-07-18:
I second this. I used that feature in Cura often. Using an array of values, instead of a fixed 90 rotation, is very nice when using designs based on triangles or hexagons. Being able to set the fixed rotation value would work too, but using arrays is way more flexible.

MikeAtBS, 2025-03-16:
+1 from me. I'm designing an elastic wheel to 3D print, and I worry that on some micro level, the fact that there are only 2 directions of infill 90 degrees apart will make the wheel's elasticity and strength act differently depending how rotated the wheel is, kinda a square wheel - so I'd rather have much more different directions of the infill to distribute the load and elasticity evenly. I guess it's do'able with modifiers to some degree but it's a pain in the ass to do it that way.

SoftFever, 2025-06-23:
Implemented in #9924

closed 2025-06-23 by SoftFever