OrcaSlicer/OrcaSlicer #3971: Unable to send job to Bambu Lab Printer's SD card

samwiseg0, 2024-02-05:
### OrcaSlicer Version

1.9.0

### OS version

13.4.1 (c) (22F770820d)

### Additional system information

Apple M1 Max

### Printer

Bambu Lab X1

### How to reproduce

1. Slice any object.
1. Click the drop down and select "Send" instead of print plate
1. Observe that OrcaSlicer never sends the file to the printer. 

### Actual results

The window hangs and never sends the Job to the printer's SD card. It says sending file over LAN but nothing happens. 

[image]

### Expected results

The file to be uploaded to the printer's SD card. 

I have tested this same method with Bambu Slicer and it works just fine. 

### Project file & Debug log uploads

[logs.zip](https://github.com/SoftFever/OrcaSlicer/files/14159081/logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

jerworrall, 2024-02-09:
seeing same behavior on windows 11

CCS86, 2024-02-13:
I'm suddenly getting this on 1.9.0 too. Windows 10. 

I checked Bambu Studio and it also did not send, but it popped up another dialog where I could input the IP address and access code, then click test. It passed the test and then could send gcode after that. Orca still cannot, despite having connectivity to the printer, including live view, SD card contents, etc.

[image]

[image]

[image]

[image]

CCS86, 2024-02-22:
@SoftFever?  

This one is a pretty big deal.

tlhintoq, 2024-02-23:
Maybe give the nightly build a try. (That would be 2.0-dev)
Its working for me.
[image]

samwiseg0, 2024-02-23:
@tlhintoq 

I tested with nightly and still have the same issue. 

Note my printer is not in lan only mode. 

[logs.zip](https://github.com/SoftFever/OrcaSlicer/files/14381591/logs.zip)

tlhintoq, 2024-02-23:
@samwiseg0 
> Note my printer is not in lan only mode.
I don't recall making any change like that to my setup.  Its pretty stock.  Did the installation per the instructions.  Got it on the LAN.  Can print to it, view the camera, blah blah blah.  Been cranking out parts ever since.
I honestly don't know what/where "lan only" mode is.  I don't think I'm there as I also commonly walk SD cards to the printer to run jobs.

samwiseg0, 2024-02-23:
Not sure why your setup is working. Hopefully someone can look into this issue when they get some time.

tlhintoq, 2024-02-23:
> Not sure why your setup is working. Hopefully someone can look into this issue when they get some time.

I notice your machine name is "X1C-86".  Nice name.  But not in the same format as the name issued to my machine automatically when I registered with the Bambu network.
[image]
Did you rename it?  I'm just wondering what other changes might exist between versions of our machines.  Firmware?

There's nothing special about configuration at the machine: Everything is really at the slicer.  Have you tried doing a total factory reset so its like new out of the box and then just not change anything?  See if its happy then?

And of course the obvious stuff like: Both networking issues.  I'm ethernet to my router then wifi to the printer, so its only 1 leg of wifi traffic: Maybe speed of transmission has something to do with it.  Or that you don't have PC and printer on the same 2.4g network if your router has the 5.0 and 2.4 on the same name as they often do.

samwiseg0, 2024-02-23:
@tlhintoq its not a printer issue since it works on bambu slicer. Secondly, I would prefer to troubleshoot based off of data and not guess as to what the case may be. I have not had a lot of time to dive into orca slicer logs and decipher what the issue may be.  

I am almost certain it is a software bug in orca slicer. Since this effects other users as confirmed above.

tlhintoq, 2024-02-23:
> I am almost certain it is a software bug in orca slicer since this effects other users as confirmed above.

Just got to figure out what the magic combination is.  The more the user-base can figure out the more likely the developers are to fix something.  They've got 500 tickets to work on.  It just stands to reason that if someone can narrow down the hunt then that ticket gets more attention.

If a few people find a pattern that helps:
It only happens when the job is bigger than 500k
It only happens on machines that got renamed
It only happens on jobs with 10 or less parts on the plate
It only happens when my PC is connected to the network over wifi but never when I'm ethernet cabled
It only happens when the SD card has less than 500mb free space
etc.

Yeah I know it feels like being free QA help for the developers - but then again the program is free - and everyone gains from everyone's contributions.

SoftFever, 2024-02-24:
I tried with my P1P machine.
I'm able to send to SD with 1.9.1 + updated plugin.

samwiseg0, 2024-02-24:
@SoftFever I had tried that before and once again now with the same result. 

Could it be something where the UI is blocking the popup or not invoking it? In bambu slicer I have a popup that appears after first trying to send a file.

SoftFever, 2024-02-24:
> @SoftFever I had tried that before and once again now with the same result.
> 
> Could it be something where the UI is blocking the popup or not invoking it? In bambu slicer I have a popup that appears after first trying to send a file. ![2024-02-24_08-34](https://private-user-images.githubusercontent.com/2241731/307525316-651cd844-3e0c-4e3d-acd6-41cba308a1c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3ODIwMTksIm5iZiI6MTcwODc4MTcxOSwicGF0aCI6Ii8yMjQxNzMxLzMwNzUyNTMxNi02NTFjZDg0NC0zZTBjLTRlM2QtYWNkNi00MWNiYTMwOGExYzMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjRUMTMzNTE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTY5NjJmZjBkOGY5ZWQwN2ZiNzA5MDMxOTAwZWY1MDA5ZTkxM2M4Mjk5ZTA5NTM1YjM3ZGZhMjI5ZmExMjJjNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Xp61ZJciOBXy8dpeFD7L5l-Nw9wEjOCNGmQFAl7Nxmk)

It's very unlikely. Most likely a network related issue.

samwiseg0, 2024-02-24:
> It's very unlikely. Most likely a network related issue.

It works fine in Bambu studio. I did the same plugin and ota wipe and tested again.

samwiseg0, 2024-02-24:
In orca I have never been presented with the second dialogue to enter my bbl user/ lan password.

samwiseg0, 2024-02-24:
Orca slicer:
https://github.com/SoftFever/OrcaSlicer/assets/2241731/195e91a0-3496-4f95-aa3f-bfc4c66b7c92

Bambu Studio:
[image]

samwiseg0, 2024-02-24:
> the dialog will only prompt when use LAN mode.
> Can you try to switch to LAN mode then switch back?

I popped up in bambu studio in non lan mode. I assume its the same plugin being used for both?

SoftFever, 2024-02-24:
> > the dialog will only prompt when use LAN mode.
> > Can you try to switch to LAN mode then switch back?
> 
> I popped up in bambu studio in non lan mode. I assume its the same plugin being used for both?

With so little info, it's hard to tell what's wrong.
Maybe the plugin confused itself when BS and Orca using different version of it.

You can try to close BS. 
Then try to delete Orca's plugin and ota folder see if it fixes the issue.

samwiseg0, 2024-02-24:
@SoftFever I will do some more investigation and possibly some packet captures to see what maybe happening. 

I will report back once I have done some more testing. 

Thank you for your time!

SoftFever, 2024-02-24:
> @SoftFever I will do some more investigation and possibly some packet captures to see what maybe happening.
> 
> I will report back once I have done some more testing.
> 
> Thank you for your time!

👍

samwiseg0, 2024-02-24:
@SoftFever After a bit more digging I discovered what that dialouge was doing in BS. It appears to write the serial and the IP of the device to config file. 

Bambu Studio:
Before dialogue
[image]
After dialogue
[image]

Now. Taking that information and placing it in the `OrcaSlicer.conf` allows OrcaSlicer to send the file to the printer.
[image]

Once my printer is converted into lan only mode it is no longer discoverable over the network via BS or OS. I am still investigating as to why as the printer and my laptop are on a flat L2 network with no multicast mitigation. That is only limited to discovery. They can still talk to each other via their IPs so this maybe a L2 discovery issue only. 

I believe this could be easily replicated if someone places their printer on a separate L2 network, still has it connected to Bambu's cloud and can still access it via its IP. The cloud at that point facilitates the discovery. 

But that said I still think there is an issue where OS does not pop the dialogue to be able to write the IP of the printer to the config file. 

Is there a way to backtrack how that is written to config file? Maybe that will help shed some light on the issue?

SoftFever, 2024-02-24:
> @SoftFever After a bit more digging I discovered what that dialouge was doing in BS. It appears to write the serial and the IP of the device to config file.
> 
> Bambu Studio: Before dialogue ![2024-02-24_10-06](https://private-user-images.githubusercontent.com/2241731/307531136-9d30cc34-ae60-4884-918f-a5334c58b307.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3ODg2MzAsIm5iZiI6MTcwODc4ODMzMCwicGF0aCI6Ii8yMjQxNzMxLzMwNzUzMTEzNi05ZDMwY2MzNC1hZTYwLTQ4ODQtOTE4Zi1hNTMzNGM1OGIzMDcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjRUMTUyNTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTViOTU5MjM1N2MzMTUwNTE3M2UxNTY3NjdmYzNiNmI5MWE4ZWYxNWNhYWY4ZWZlYWM3ZDIyMmMwYzdiYTI5YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yIRdUyHffxE8cQoF3INnF5p725dF97hSDUk-pmNmwD0) After dialogue ![2024-02-24_10-07](https://private-user-images.githubusercontent.com/2241731/307531162-423e2e63-0ef2-4a44-ad52-abf30e5bd623.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3ODg2MzAsIm5iZiI6MTcwODc4ODMzMCwicGF0aCI6Ii8yMjQxNzMxLzMwNzUzMTE2Mi00MjNlMmU2My0wZWYyLTRhNDQtYWQ1Mi1hYmYzMGU1YmQ2MjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjRUMTUyNTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWJmODUzMDc2NWQyYTQ1MjNhMmExYTM2MDVlODgwMTA1OGE2MWMzY2FhOTBhMmJmZWYzN2M5OGIzOTQzYjU5MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.c9pSRa5K8oS1NqQExEzpJiyEZdTOWojUXLwmh37n6qg)
> 
> Now. Taking that information and placing it in the `OrcaSlicer.conf` allows OrcaSlicer to send the file to the printer. ![2024-02-24_10-09](https://private-user-images.githubusercontent.com/2241731/307531243-559c926e-bd2f-4e58-84b0-751b645bc175.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3ODg2MzAsIm5iZiI6MTcwODc4ODMzMCwicGF0aCI6Ii8yMjQxNzMxLzMwNzUzMTI0My01NTljOTI2ZS1iZDJmLTRlNTgtODRiMC03NTFiNjQ1YmMxNzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjRUMTUyNTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGM3ZTQ2MGNlN2FhNzc5ZDBlMzQyYjMxYjEwOWYyMTM1NDUxZGJhOGFjMDhhMWE3MjI2MWFhNDlmNjk3YWQwYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.az-8ER18ywZGTk4OqiTNTyjCooWF0dr2yrsMWwP0n-g)
> 
> Once my printer is converted into lan only mode it is no longer discoverable over the network via BS or OS. I am still investigating as to why as the printer and my laptop are on a flat L2 network with no multicast mitigation. That is only limited to discovery. They can still talk to each other via their IPs so this maybe a L2 discovery issue only.
> 
> I believe this could be easily replicated if someone places their printer on a separate L2 network, still has it connected to Bambu's cloud, and can still access it via its IP. The cloud at that point facilitates the discovery.
> 
> But that said I still think there is an issue where OS does not pop the dialogue to be able to write the IP of the printer to the config file.
> 
> Is there a way to backtrack how that is written to config file? Maybe that will help shed some light on the issue?

This is helpful.
The dialog is in this function
`bool GUI_App::show_modal_ip_address_enter_dialog(wxString title)`

It is supposed to be triggered if send job failed.
Unclear why it didn't trigger
```
    m_send_job->on_check_ip_address_fail([this](int result) {
        wxCommandEvent* evt = new wxCommandEvent(EVT_CLEAR_IPADDRESS);
        wxQueueEvent(this, evt);
        wxGetApp().show_ip_address_enter_dialog();
    });
```

cadriel, 2024-02-24:
fwiw - I've been having this problem. My X1C is cloud only setup - and I've never had that dialog show. I opened my `orcaslicer.conf` and added the entry for the IP address (my machine is set to a fixed IP) and I can now send gcode files without issue. I've seen no other adverse affects as yet.

witchrobin, 2024-06-18:
I am also seeing the exact problem with OS when sending to sdcard.  First time I try to send, it will say sending via clould but doesnt' actually do anything. If I cancel and send again, it will say sending via LAN and also sits there. 

Same test works within BS where it will pop up with a diaglog box to let me enter the IP address of my printer and then it start the transfer. My printer is NOT in LAN only mode and my PC  and P1S are on different VLAN. I am able to ping P1S from PC.

I also tried modifying the orcaslicer.conf to put in the ip_address json object, but it doesn't seem to stick. As soon as I launch OS, that block gets deleted from the conf file.

CCS86, 2024-06-18:
Yes, this is still an issue. 

My printer recently grabbed a different IP address from the router and Orca Slicer was lost. It can see the printer, and even "print" jobs, but it cannot send to SD card. It lacks the pop-up in BS that lets you change the IP and LAN code and test.

npopson, 2024-07-02:
I started having this issue several months ago (exactly as described by @samwiseg0) and decided to look into it further today. I am currently on OrcaSlicer 2.1.1.

- Deleting 'plugins' and 'ota' folders had no effect.
- BambuStudio correctly prompted for the IP address and is able to send prints without issue.
- Copying the "ip_address" block from BambuStudio.conf to OrcaSlicer.conf resolves the issue (thanks @samwiseg0 !). 

Additional notes if it helps identifying the bug: 
- My OrcaSlicer.conf we missing the entire "ip_address" block.
- If intentionally set an incorrect IP address and try sending: BS will prompt to fix the IP through the GUI. OS does not prompt and gives a "Error Code: -5010" Failed to upload print file to FTP.

Nitroxid, 2024-07-24:
I'm having the same issue, but only on one of my PCs.

My Desktop-PC has the issue, my Laptop doesn't. Both have OS 2.1.1 on Win 10.
Both OrcaSlicer.conf file basically look the same. Some differences in the presets, but I doubt this will have an effect.

My Desktop is connected via LAN to a router which acts as a Wifi-Bridge to the Router. 
My Laptop and the X1C are connected to the Router directly via Wifi.

If I try to send a job to the printer, it fails with error code 0. It does this when trying to send via cloud service and via LAN. Directly printing is working though.

When I add the IP address to the OrcaSlicer.conf on my Desktop it is working.

doudar, 2024-08-19:
I completely re-installed windows on this machine (due unrelated hard drive failure), and Orcaslicer still doesn't send to the SD.

doudar, 2024-08-19:
Okay. I was able to get this working. Steps:

1. Go into my router and assign a static IP to the X1C.
2. Set the new IP address for the X1c in  ~AppData/Roaming/OrcaSlicer/OrcaSlicer.conf
3. On the front panel of the X1C, generate a new access code by clicking the circle icon next to the current code.
4. Open Bambu Studio, try to send a file, and enter the new access code in the box that pops up.

I tried all of the other variations posted, and none of them worked for me. This did the trick.

pontos-dev, 2024-08-26:
After I have formatet computer and had to install orca slicer again (2.1.1) I had the same issue with my bambu lab x1 carbon, it could'nt send a file to the printer, while bambu studio could with a problem
After reaading this issue here, I tried to find my bambustudio.conf and orcaslicer.conf, and as some one has stated before, the whole ip section of the .conf file is missing, and after I added that, it just worked

Hope you guys (SoftFever) finds out why this is happening, I have never had issues with OrcaSlicer before this issue

yummypho, 2024-08-29:
> After I have formatet computer and had to install orca slicer again (2.1.1) I had the same issue with my bambu lab x1 carbon, it could'nt send a file to the printer, while bambu studio could with a [...]

Just adding a datapoint that the above fix worked for me.   I ran into the same issue with "Send" not working in OS but BS worked fine.    Comparing the bambustudio.conf and orcaslicer.conf to find and add the Serial + IP fixed it for me.

WimVDK, 2024-10-04:
Using OC 2.2 beta 2, this error still exists. Especially after swapping out the X1C AP board (which gives the printer a new serial) and you rename the printer to the same name the old one had

DragonFlyer007, 2024-10-12:
Turn of your windows firewall and try again. Restart bambu studio first.
If it works now, check that your settings, that its private network and bambu is allowed thru the firewall in private network also.
Also check for duplicate of bambu in the firewall. Delete one if two.

This solved it for me.
Fresh Win 11 install and all latest updates and firmware.

WimVDK, 2024-10-12:
Worked fine for a year, didn't change anything. It's not my windows or AV/Firewall cause the standard 'print plate' works fine which would need to pass through the same firewall

emollusion, 2024-12-01:
I stubmled upon this when I put my printer behind a firewall, being proprietary and all that.
It is allowed to access the internet, so the cloud service work.
But the Orcaslicer is not able to send the file.
At the beginning I got the FTP error for port 990 from the print (I presume?) to my computer running orca. Allthough I did not see that traffic in the FW logs I did put a rule to allow it. Did not change anything.
I thenf ound this, so I went ahead and entered a IP address in the conf and did not take a backup for some reason, so when I started the slicer again it said the conf was broken, overwrote it.
I now have printer online in the slicer, and in the cloud service, and I can send prints from the cloud service, but not from the slicer, but now the error code is just "0", no description or info.

rshev, 2024-12-18:
Still an issue on OrcaSlicer 2.2.0 on Mac with A1 NOT in LAN mode. BambuStudio works fine.

Bass3d, 2025-02-10:
I had the same issue, with all newer Orcaslicer updates. I just made it working. In your Windows Firewall delete all occurrences of OrcaSlicer, and make a new one, with Private Networks permission flagged. It worked for me, now I can send files from pc to Bambu Lab A1 mini through my local net. Hope it helps. Stefano

closed 2025-05-20 by github-actions[bot]