OrcaSlicer/OrcaSlicer #3973: Enable Filament Change at Layer When Print Sequence is By Object

qeggs, 2024-02-05:
Default print sequence is "By Layer", with the option to switch to "By Object". 

During typical prints sequenced "by layer", it is convenient to add a filament change, or a pause or custom G-code at a particular layer after the part is sliced.

These same features are not available (grayed out) in prints sequenced "by object". It would be nice to have the change filament option available at least.

The alternative that I'm using right now is to put my objects on multiple plates so they don't need to be sequenced "by object". This workaround allows me to specify filament changes.

I'm using a Bambu X1C.

[image]

mcgeld, 2024-06-25:
Just ran into this and was disappointed to see that I couldn't add filament changes in a print by object print.

ntroutman, 2024-08-26:
This appears to be for the other options as well, such as adding a pause. I just ran into this today, and had to do the same thing of breaking the one object out to its own file by layer.

jeanlescure, 2024-09-07:
I also need this feature, it would allow me to save 9min between each object print.

To give better context, I've run a stopwatch to time how long it takes between the end of one print and the start of the first layer on the next print. It takes about 11min, of which only 2min are bed leveling and purge line, the rest is pretty much bed/nozzle heating up (I use high temp. filaments).

What I want to do is have 9 copies of my object being printed in sequence, but add a pause and custom gcode whenever it finishes printing an object.

The pause would allow me to just snap the latest print off of the build plate as well as pull off the purge line, then after the pause the custom gcode would run bed leveling and purge line again right before starting on the next object. No waiting for bed/nozzle heating.

Just by being able to do this I would potentially save pretty much 81min of wait time for every 9 of these objects that I print!

I hope this example helps you visualize how this feature can be extremely valuable for us users :)

nate1mace, 2024-11-20:
Bump this would be great. Would also like to control filament sequence by layer - not just first layer.

iamstickfigure, 2024-12-09:
Bump. I just ran into this problem as well. 

I wanted to do a PLA print with a filament change on a specific layer, but my last print was PETG, and I figured whatever my next print was might not have great layer adhesion. So I figured I would do a "by object" print and have the first item be something that doesn't need to be strong. Then once it got to the other item, the PETG would be flushed out by that point.

But since "by object" removes the ability to do a filament change, I'll have to do a separate print first, which will take more of my time 😕

closed 2025-03-17 by github-actions[bot]