OrcaSlicer/OrcaSlicer #3977: Extrusion Multiplier for Each Feature Type

arshish1612, 2024-02-05:
I originally created this issue: https://github.com/SoftFever/OrcaSlicer/issues/1558 but it has been closed without being implemented.

Apologies for the duplicate however I cannot re-open an existing issue hence I am creating a new one here. 

---------------

_Is your feature request related to a problem? Please describe.
Unable to control flow rates / extrusion multipliers for different feature types. Currently you can only adjust the default multiplier and the top Solid Infill. However there is a need to set the flow rate for different layers. This is available in Cure and is extremely useful in getting prints perfectly dialled in.

Describe the solution you'd like
Using Cura as a reference, a section to se the extrusion multiplier for different line features.
[image]

Describe alternatives you've considered
In PrusaSlicer you have a substitute GCode which is OK but Cura implements it better

Additional context
Add any other context or screenshots about the feature request here._

closed 2024-05-13 by github-actions[bot]

Erosion139, 2025-03-23:
This is wild, whats with the complete ghosting over this very important and useful feature? I am trying to make a profile for the 'PETG glass" printing and cannot set a separate infil multiplier to fill the gaps properly. Huge bummer, guess ill use Cura.

3DFlavours, 2025-04-06:
I'm also blown away, that this issue has been ignored. Cura is the best when tuning flow. Have been trying to dial in dimensionally accurate prints with varying necessity for flow on many areas, and just can't with Orca. Feels like I will just have to dust off Cura again. Too bad as I love the UI and printer and filament organization of Orca.