OrcaSlicer/OrcaSlicer #3983: Printing by object GCode does not get applied between objects

ruddy17, 2024-02-05:
### OrcaSlicer Version

1.9.0

### OS version

Win 11

### Printer

Voron 2.4

### How to reproduce

So, while printing sequentially I need to raise Z a bit to avoid knocking printed parts off. I open printer settings and add this to 'Printing by object gcode' field:
```
;RAISE_Z_BEFORE_NEXT_OBJECT
G91
G1 Z5
G90
```
Then save settings and hit 'slice'

### Actual results

The gcode is not actually inserted between objects, could be verified by searching for ;RAISE_Z_BEFORE_NEXT_OBJECT in the resulting gcode file. Also, the expected travel move is not shown on preview. 

### Expected results

Obvioulsy, the gcode should be inserted between each of the object instances. 

### Project file & Debug log uploads

[Calibrator(4).zip](https://github.com/SoftFever/OrcaSlicer/files/14166240/Calibrator.4.zip)
plates 5 and 6 are affected. here is also the resulting gcode I get:
[Uploading Calibrator(4)_PLA_1h14m.zip…]()
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14166269/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Meth0d007, 2024-02-05:
mabye this ist related to the z-hop issue https://github.com/SoftFever/OrcaSlicer/issues/3747

ruddy17, 2024-02-05:
@Meth0d007 don't think so. My issue is not about z-hop at all, but rather about failure to add any custom gcode between objects with sequential (by-object) printing mode

closed 2024-05-13 by github-actions[bot]