OrcaSlicer/OrcaSlicer #3984: Crash on profile save

SteynMarnus, 2024-02-05:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Additional system information

AMD Ryzen 7 3700X
Gigabyte NVIDIA RTX 3070Ti OC Vision Edition
G-Skill F4-3600C16 RAM

### Printer

Creality K1

### How to reproduce

1. Click on "Save current process"
2. Enter new name for adjusted changes
3. Click "OK"

### Actual results

Application Hangs/Crashed. See attached recording

https://github.com/SoftFever/OrcaSlicer/assets/19153368/ae337c8b-4de9-4f70-a85f-c8e5003df009

### Expected results

Process to be saved successfully

### Project file & Debug log uploads

[Helios 44-2 housing_Front_adjusted.zip](https://github.com/SoftFever/OrcaSlicer/files/14167851/Helios.44-2.housing_Front_adjusted.zip)
[debug_Mon_Feb_05_17_36_00_18968.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14167862/debug_Mon_Feb_05_17_36_00_18968.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MrBigBear, 2024-02-06:
I am also experiencing the same crash. 

Orca will crash if trying to save the process with a different name. If saving with the default name (unmodified name) then Orca will save normally.

yohanson, 2024-02-08:
Same here on Windows 10.

Rulevoid, 2024-02-12:
Same issue here. Tried 2.0.0-dev but can't figure out how to get all my profiles/settings merged in.

stefanofinetti, 2024-02-19:
Same issue present on 1.9.1 both on Windows 11 and MacOS (Apple Silicon Chip)

MantisCa, 2024-02-21:
Same issue on 1.9. Thought maybe it was a permissions issue and tried upgrading to 1.9.1. Upgrade worked but still crashes when I try to save the profile settings with a new name. Running on Windows 10.

deliciousllc, 2024-03-03:
Confirming same issue. I ran through a bunch of different names to see if it was an issue with hidden or specific characters. No luck. I was able to characters to the existing name and save the profile, but changing the name causes a crash.

avxmw, 2024-03-03:
Same issue on 1.9.1 Linux AppImage. Made modifications and wanted to save as "xxx 0.2" (original profile name is "xxx 0.3"), crashed. Tried again, crashed again.

stefanofinetti, 2024-03-03:
I've managed to save new profiles, but it requires to manually edit the json files in /appData/Roaming/OrcaSlicer/user directories.
After renaming the profile I was using removing the @MyMarlin (or @MyKlipper) it started saving with new names. So apparently it's just a problem of the "@" in the name that is not correctly escaped. I just finished saving my different custom profiles for .2 layer height for PLA and PETG. So far so good.

closed 2024-06-09 by github-actions[bot]