OrcaSlicer/OrcaSlicer #3985: Disable Gap Infill

Stargo44, 2024-02-05:
Since the release 1.8, we are not able to set the speed of "Gap Infill" to 0.
It was used to get rid of the "Gap Infill". The only method that I know of.

Every printer will benefit of this feature.

The solution would be to set a paramter (On/Off) to disable or not the Gap Infill (as prusa slicer have).

On this picture you can see a scenario where I don't want to have the Gap Infill. 
[image]

dartrax, 2024-02-07:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/2414

Stargo44, 2024-02-07:
> Duplicate of #2414

It's working for tiny gap but not for all gap as the image below. Watever threashold I set for the "Filter ou tiny gaps" setting.

[image]

A check box to remove everithing would be great as Prusa Slicer.

Stargo44, 2024-02-07:
It's working with the setting "Filter ou tiny gaps", I didn't know.
Sorry for the post.

closed 2024-02-07 by Stargo44

cubitik, 2024-08-13:
If anyone wonders how to turn it off, here is the cure: Go to the Strength tab, find the setting "Apply gap fill". Choose an option other than nowhere. Now there should be another setting visible "Filter out tiny gaps", now set it to 9999999 (whatever highest value you can) and ta da! It won't fill again.