OrcaSlicer/OrcaSlicer #3987: Orca Slicer doesn't Detect Bambu X1 Carbon

Marthisdil, 2024-02-05:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Bambu Lab x1 Carbon

### How to reproduce

Go to devices, don't see printer.

Bambu Studio can see it just fine

### Actual results

Not able to see printer

### Expected results

See printer

### Project file & Debug log uploads

[card_divider_vertical.zip](https://github.com/SoftFever/OrcaSlicer/files/14172151/card_divider_vertical.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-02-08:
FYI:  Version 2.0, nightly build from 06feb2024 on Windows10 - is seeing my X1C... Its sending jobs to it and getting feed back in real time.
@Marthisdil  - Maybe try the version 2 to rule out if its a version issue or something odd in Win11 ?

camled, 2024-02-19:
It also does not detect Bambu P1S, Orca Slicer v1.9.1

closed 2024-05-27 by github-actions[bot]