OrcaSlicer/OrcaSlicer #3989: Crashing when trying to view P1S camera

ZephyFoxy, 2024-02-05:
### OrcaSlicer Version

1.9.0

### OS version

Kali Linux 2023.4/Debian 12

### Additional system information

CPU: AMD Ryzen 5 5600X
GPU: NVIDIA Geforce RTX 3050
DE: Plasma 5.27.10

### Printer

Bambu Labs P1S

### How to reproduce

Go to 'Devices' tab
Select P1S

### Actual results

Application crashes while trying to load the camera feed for the P1S, but loads X1C camera fine. This issue was not present on the Windows version.

### Expected results

Should see camera feed.

### Project file & Debug log uploads

[Log.zip](https://github.com/SoftFever/OrcaSlicer/files/14172712/Log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

belzebub40k, 2024-02-20:
I have the same problem with OrcaSlicer and BambuStudio. Bambu Labs provides an additional AppImage for Fedora using an older libjpeg version. The version of BambuStudio using the older libjpeg does not crash for me on Debian 12. If something similar could be done for OrcaSlicer this would be great.

https://github.com/bambulab/BambuStudio/issues/1665#issuecomment-1535569629

tobiasium, 2024-06-07:
orcaslicer (all versions I tried) crashes on Debian 12 when trying to view the camera. On the command line "segmentation fault" is shown.

closed 2024-09-13 by github-actions[bot]

jgoguen, 2025-02-23:
This remains broken on Fedora 41 Plasma, using Orca Slicer 2.2.0 AppImage downloaded from the Releases page. I also have a X1C, same network same account, works perfectly fine.

OS: Fedora 41, KDE Plasma
CPU: 13th Gen Intel(R) Core(TM) i9-13980HX
Memory: 64GB
GPU: GeForce RTX 4090
Printer: Bambu Lab P1S

My log files are basically non-existent though. Starting from Orca closed and an empty log directory, after Orca loads there's only two files: an encrypted network debug log and the application debug log. The entire log file, which is printed before I attempted to do anything at all, is:

```
[warning]   2025-02-22 19:15:25.679185[Thread 0x00007f0c090dd740]:get_network_function, can not find function bambu_network_del_subscribe
[warning]   2025-02-22 19:15:25.679222[Thread 0x00007f0c090dd740]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]   2025-02-22 19:15:25.679231[Thread 0x00007f0c090dd740]:get_network_function, can not find function bambu_network_get_profile_3mf
[warning]   2025-02-22 19:15:25.850025[Thread 0x00007f0c090dd740]:Preset already present, not loading: eSUN PETG Basic @Bambu Lab X1 Carbon 0.4 nozzle
[warning]   2025-02-22 19:15:25.850035[Thread 0x00007f0c090dd740]:Preset already present, not loading: eSUN PETG Basic @Creality Ender-3 V3 Plus 0.4 nozzle
[error] 2025-02-22 19:15:26.394503[Thread 0x00007f0b747f86c0]:Unable to open the file/home/jgoguen/Downloads/Pixel+Heart+.3mf
[error] 2025-02-22 19:15:26.394501[Thread 0x00007f0b7c5fc6c0]:Unable to open the file/home/jgoguen/Downloads/lovebear.3mf
[error] 2025-02-22 19:15:26.394539[Thread 0x00007f0c090dd740]:Unable to open the file/home/jgoguen/Downloads/Yoda+Only+One.3mf
[error] 2025-02-22 19:15:26.394559[Thread 0x00007f0b307f86c0]:Unable to open the file/home/jgoguen/Downloads/ToolBox-Large.3mf
[error] 2025-02-22 19:15:26.394611[Thread 0x00007f0b747f86c0]:Unable to open the file/home/jgoguen/Downloads/smb_item_block_lid+BBS+PLA.3mf
[error] 2025-02-22 19:15:26.394607[Thread 0x00007f0b107f86c0]:Unable to open the file/home/jgoguen/Downloads/SmallPartsStorageBox_RevB.3mf
[warning]   2025-02-22 19:15:26.611240[Thread 0x00007f0c090dd740]:07:15:26 PM: Warning: Error running JavaScript: Unsupported result type
[warning]   2025-02-22 19:15:26.612340[Thread 0x00007f0c090dd740]:07:15:26 PM: Warning: Error running JavaScript: Unsupported result type
[warning]   2025-02-22 19:15:26.613336[Thread 0x00007f0c090dd740]:07:15:26 PM: Warning: Error running JavaScript: Unsupported result type
[warning]   2025-02-22 19:15:26.645057[Thread 0x00007f0c090dd740]:07:15:26 PM: Warning: Error running JavaScript: Unsupported result type
```

Nothing at all is printed after this. I click on Devices, select the P1S (if it isn't already selected), click Play, and after a few seconds Orca crashes. No additional logs are printed, and the network log file is not changed.