OrcaSlicer/OrcaSlicer #3990: Make X-Y hole compensation a filament specific setting

HaernCalistinson, 2024-02-06:
**Is your feature request related to a problem? Please describe.**

When printing with different filament (and more frequently filament types, abs, asa, etc) I've found that the xy hole compensation is not static across the board. Making this the option of being a filament specific override would make keeping track of what compensation is needed much easier. This would be critical for multi-color / multi material prints where different filaments need different compensation values.

**Which printers will be beneficial to this feature**
Klipper based (all printers?)

**Describe the solution you'd like**

Make a filament specific override option for XY Hole Compensation

**Describe alternatives you've considered**

None

tlhintoq, 2024-02-08:
Wouldn't same spool on two different machines also have different values?
The same eSun Black PETG on and ender3 behaves differently than a neptune and is different than on an X1C.
Not to mention different between a .4 and .6 nozzle.  So is it really the filament?  Or perhaps the print profile for that material on that machine?

HaernCalistinson, 2024-02-12:
> Wouldn't same spool on two different machines also have different values? The same eSun Black PETG on and ender3 behaves differently than a neptune and is different than on an X1C. Not to mention [...]

What would actually make more sense is to have different sets of profiles for each printer. You select a new printer, you start with the basic set of "generic" filaments and have to build out your profiles again, switch printers, and the only profiles to show are the ones that were configured when that printer was the actively selected one.

Setting up different profiles for different nozzle sizes is quite common. So in this case, it is the filament in the context of all the other variables (nozzle, printer, etc.).

Ultimately, if I have two filaments on the same printer with the same print profile and one of them needs a different setting than the other to print accurately, in my eyes that makes it a filament specific setting.

tlhintoq, 2024-02-12:
Pretty sure you just repeated what I said but in more words:
> Or perhaps the print profile for that material on that machine?

I think we're both saying the same thing: Per material, per machine.

Glad to see at least two people see it this way

closed 2024-05-21 by github-actions[bot]