OrcaSlicer/OrcaSlicer #3991: Custom Filament & Filament Profiles Disappearing

CryPt00n, 2024-02-06:
### OrcaSlicer Version

1.9.0

### OS version

Windows

### Printer

Bambu Lab X1C

### How to reproduce

1. Create Custom Filament Profile
2. Do prints, calibration stuff, etc.
3. At some time, Orca slicer crashed

### Actual results

Custom Filament and all created from this one are disappearing. After manually re-creating, calibration values are still there. Also, files are still in the folders. I lost multiple profiles now due to this.

Additionally, also the history of my printed files is lost, looks like multiple days are gone.

### Expected results

Custom Profile stays and does not disappear

### Project file & Debug log uploads

[debug_Tue_Feb_06_01_46_46_9612.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14173513/debug_Tue_Feb_06_01_46_46_9612.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

kylek29, 2024-02-06:
Do you have Sync Profiles turned on in preferences? I used to have disappearing profile/settings when I had that enabled. It's an issue that seems to pop-up on here and disabling that was my workaround -- I then handle syncing via Google Drive (symlinked the appropriate folder to a sync'd folder).

CryPt00n, 2024-03-07:
Yeah this was enabled for me. I'll disable, but no issues so far until now

CryPt00n, 2024-03-16:
Saw in the meanwhile, that this setting is useful when using custom filament. Without sync enabled, custom filaments will not be synced to the printer (keyword PA values).

caste1200, 2024-04-10:
I have this issue were sometimes I close orca and when I open it again, all my user presets are gone. Only system presets are left. 
Then I close Orca again and when I open it back, the presets are back.
I already disabled sync presets on preferences but no change.

krissrex, 2024-04-26:
This happens to me when I click "Set filaments to use", remove one I'm not using (like Anycubic Generic ASA). After clicking confirm, all but one user preset dissapears, and all my processes dissapear.

If I then close OrcaSlicer fully, open it again, and load one of my old saved projects (that uses one of the gone profiles), all my profiles load back in.

caste1200, 2024-05-10:
> I have this issue were sometimes I close orca and when I open it again, all my user presets are gone. Only system presets are left. Then I close Orca again and when I open it back, the presets are back. I already disabled sync presets on preferences but no change.

I think Orca somehow is loading 2 different "profiles", and each time I start orca it opens a different profile with no presets and next start it opens the profile with all the presets. Where are orca slicer profiles stored? or maybe a file that tells it to load some profile that doesn't exist anymore?

krissrex, 2024-07-17:
> Where are orca slicer profiles stored?

On windows: `AppData\Roaming\OrcaSlicer\user\default\` . You can easily open appdata by Windws+R and typing `%APPDATA%`.

closed 2024-10-23 by github-actions[bot]