OrcaSlicer/OrcaSlicer #3994: pa calibration "PA Pattern" generates object with size of the print bed

lynara, 2024-02-06:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Voron 2.4

### How to reproduce

Select Calibration, Pressure advance, select PA Pattern

### Actual results

Object generated / exclude object size in gcode is the size of print bed -5.
Adaptive meshing probes the wohle print bed.

### Expected results

Object generated / exclude object size in gcode should reflect the actual size of the pattern print.

### Project file & Debug log uploads

should not be needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JamesH1978, 2024-02-12:
I can confirm this is the same for me, my printer just tried to probe the probe dock...

JamesH1978, 2024-02-12:
`EXCLUDE_OBJECT_DEFINE NAME=Orca-PA-Calibration-Test CENTER=0,0 POLYGON=[[5,5],[230,5],[230,230],[5,230],[5,5]]` my center is not 0,0

Also sliced something else and it comes out fine, with the proper CENTER:

`EXCLUDE_OBJECT_DEFINE NAME=Raspberry_Pi_Wall_Mount.STL_id_0_copy_0 CENTER=117.5,117.5 POLYGON=[[82.4544,71.165],[82.4763,70.1743],[82.661,69.2007],[83.0034,68.2708],[83.4942,67.4099],[84.12,66.6416],[84.4434,66.3568],[84.8637,65.9867],[85.7051,65.4632],[86.6212,65.0853],[87.5869,64.8633],[88.5761,64.8034],[146.424,64.8034],[147.413,64.8633],[148.379,65.0853],[149.295,65.4632],[150.136,65.9867],[150.88,66.6416],[151.506,67.4099],[151.997,68.2708],[152.339,69.2007],[152.524,70.1743],[152.546,71.165],[152.546,163.835],[152.524,164.826],[152.339,165.799],[151.997,166.729],[151.506,167.59],[150.88,168.358],[150.136,169.013],[149.295,169.537],[148.379,169.915],[147.841,170.038],[147.413,170.137],[146.424,170.197],[88.5761,170.197],[87.5869,170.137],[87.1592,170.038],[86.6212,169.915],[85.7051,169.537],[84.8637,169.013],[84.4434,168.643],[84.12,168.358],[83.4942,167.59],[83.0034,166.729],[82.661,165.799],[82.4763,164.826],[82.4544,163.835],[82.4544,71.165]]` 

This seems to be some bug in the PA calibration at least

lhndo, 2024-02-12:
Same here. I can confirm this issue.

closed 2024-05-21 by github-actions[bot]

staal54a, 2024-10-05:
I also see this issue