OrcaSlicer/OrcaSlicer #3997: z height in world coordinates does not allow you to move a item up.

NBF1998, 2024-02-06:
### OrcaSlicer Version

1.9.0

### OS version

windows 11

### Printer

bambu lab a1

### How to reproduce

import a new object on build plate, then try to move z coordinates of part.

### Actual results

you type in a new number then it resets to previous number . if you drag the part up it just goes back to build plate.

### Expected results

the z height above the build plate should stay where i set it.

### Project file & Debug log uploads

[polysoup.zip](https://github.com/SoftFever/OrcaSlicer/files/14184987/polysoup.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NBF1998, 2024-02-06:
I have to use creailty slicer anytime i want to put a logo on a print because of these issues.

gasprints, 2024-03-12:
you need to "add part" to combine them and be able to move one part in relation to the other and have the height stay
[image]

IcarianOM, 2024-04-12:
I think i have a similar issue:

[image]

if i move the OTHER part to the Mirrored Coordinates, after i set the 3 values, but y=45, the piece moved as the following image:

[image]

NBF1998, 2024-04-13:
Why is that a problem though? I imagine itd be a very easy bug for them to
fix

closed 2024-07-21 by github-actions[bot]