OrcaSlicer/OrcaSlicer #4000: Critical conflict between drawn supports and shrinkage compensation

flugegegeheimen, 2024-02-07:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Additional system information

i7 9900k
titan rtx
ram 32

### Printer

any printer

### How to reproduce

1) enable support, any manual mode, snug style for example
2) draw support on any inclined surface
3) shrinkage 100 in plastic profile
4) slice and see
5) change shrinkage to any instead 100 (99,4 for example)
6) slice and see

### Actual results

(sorry, I'm writing via Google Translate)
For a long time I tried to understand the reason for the defects on my printouts made from materials with high shrinkage...

An illustration of how the supports were drawn by hand and what I expected to see:

[image]

If you change the compensation value instead of 100% (for example, to 99.4 as in my screenshot), then the supports are built incorrectly:

[image]

I received a lot of defects because of this bug.

### Expected results

If you slice the g-code without using shrinkage compensation in the plastic profile, then everything works as normal:

[image]

### Project file & Debug log uploads

[test_shrinkage.zip](https://github.com/SoftFever/OrcaSlicer/files/14190862/test_shrinkage.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

atl285, 2024-02-07:
@flugegegeheimen, nice find. I often had problems with manual support and use mainly filament with shrinkage compensation. But I never thought, that this can be a reason for this. 😮

closed 2024-05-15 by github-actions[bot]

atl285, 2024-05-15:
Is it solved?