OrcaSlicer/OrcaSlicer #4001: Resize of object shows 100% after the size is changed

apyarnold, 2024-02-07:
### OrcaSlicer Version

2.0.0-dev_Portable

### OS version

Windows 11 Pro 64-bit 22H2

### Additional system information

CPU: 11th Gen Intel(R) Core(TM) i7-11370H @ 3.30GHz   3.30 GHz
RAM: 32.0 GB
Video: NVIDIA GeForce RTX 3050 Ti Laptop GPU

### Printer

Creality Ender 3 S1 Pro

### How to reproduce

1. Place any two .stl models on the plate
2. Select more than one model
3. Press the Scale button
4. Change any of the Scales

### Actual results

When you change to percentage to either a positive or negative eg. 98% or 120% . When you click outside the box or press enter the size changes but the percentage reverts to 100%.

### Expected results

Expected result would be the percentage reflect what was entered as happens whenever a single model or one of multiple models is selected and scaled.
[ScreenShots.zip](https://github.com/SoftFever/OrcaSlicer/files/14192974/ScreenShots.zip)

### Project file & Debug log uploads

Project File: 
[Project.zip](https://github.com/SoftFever/OrcaSlicer/files/14192992/Project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

hiltpatr, 2024-05-15:
I too have found this to be an issue. I want to scale based on the original model size, not based on the current model size. Many times I am scaling to try and fine tune dimensions without having to rebuild the model each time. It is annoying to have to undo or reload the model to scale where I want it to be.

Chix9090, 2025-04-20:
I find this very annoying too.. please fix this!