OrcaSlicer/OrcaSlicer #4002: OrcaSlicer 1.9.0 error displaying round build plate (config for FLSUN SR)

ekatus, 2024-02-07:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 22.04

### Additional system information

Intel NUC8i5BEH 16GB

### Printer

FLSUN SR

### How to reproduce

Configure a Delta Printer

### Actual results

The Build plate display is duplicated with overlay
[image]

### Expected results

Same view as 1.8.1 which is correct. 
[image]

### Project file & Debug log uploads

No configuration to show. It's obvious
[ObjectiveTube-short.zip](https://github.com/SoftFever/OrcaSlicer/files/14194264/ObjectiveTube-short.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Cascadier1, 2024-02-25:
Same here if i try to add a Delta Printer

Funny thing: Not visible in Dark Mode.
[image]

faith-healer, 2024-02-27:
same here
1.9,.x
latest 2.0.0. alpha too

[image]

closed 2024-06-06 by github-actions[bot]