OrcaSlicer/OrcaSlicer #4003: Allow iron the n'th top layers

ciroanacleto, 2024-02-07:
**Is your feature request related to a problem? Please describe.**
Nowadays the slicer has the option to iron "topmost surface only" but when using bed mesh calibration this can cause some parts of the surface has slightly difference in heigths exposing the layer below the topmost ironed surface.

[image]

There is the option to iron all top layers in the slicer but not always this is needed and is very time consuming for complex objects.

**Which printers will be beneficial to this feature**
Klipper based printer or any printer with automatic bed mesh

**Describe the solution you'd like**
It will be great if we could have the option to specify how many top layers we would like to ironing, for example the topmost 2 layers.

**Describe alternatives you've considered**
I don't know if the slicer allows change the deep hight in the layer being ironed. I believe this could help a litte bit mitigate possible differences in bed mesh hights too.
Or even allow change the direction of the ironing line to not be concident with the direction of monotonic pattern line (or similar line patterns).

**Additional context**
In the picture attached uppon, the top pattern line choosed was monotonic and the ironing was made in the same direction. Layer hight of 0.28mm with 0.6mm nozzle. Printed with PLA on Artillery sidewinder X2 printer with klipper.

Vantage-DS, 2024-02-07:
This can be achieved already using a height range modifier for the layers you want ironing -

[image]

ciroanacleto, 2024-02-07:
I did not know this feature. Indeed the behavior I have described can be achived by height range modifier. It is a little bit trickier but it is possible. I believe include a new field for ironing configuration would be more explicit and easy to understand. Thank you by the tip!

closed 2024-05-15 by github-actions[bot]