OrcaSlicer/OrcaSlicer #4004: Report of Stringing Issue in Orca Slicer

modaali96, 2024-02-07:
### OrcaSlicer Version

2.0.0 dev

### OS version

windows 11

### Additional system information

core i7 gen5 
amd r5
16 ram

### Printer

CR-10 V2

### How to reproduce

To reproduce the stringing issue in the Orca Slicer program, follow these detailed steps:

Open the Orca Slicer program on your computer.
Ensure that you have a 3D printer with a remote extruder (not a direct extruder) connected and properly configured.
Load a 3D model file into the Orca Slicer program for slicing.
Configure the slicing settings according to your preferences, including layer height, infill density, print speed, and retraction settings.
Pay special attention to the retraction settings, including retraction distance and speed.
Generate the G-code for the sliced model.
Transfer the generated G-code to your 3D printer.
Start the printing process and observe the extrusion behavior closely.
Note any instances of stringing (unwanted filament strands between printed parts) during the printing process.
Repeat the printing process with variations in the retraction settings to see if the stringing issue persists or improves.
Ensure to document any specific configurations or observations during the printing process that might be relevant to diagnosing and resolving the stringing issue.

### Actual results

After following the steps outlined above, the actual results of the stringing issue in the Orca Slicer program with a remote extruder 3D printer are as follows:

During the printing process, regardless of adjustments made to the retraction settings (including retraction distance and speed), stringing occurs consistently between printed parts.
The stringing manifests as thin strands of filament spanning across gaps between printed sections, resulting in unwanted material buildup and poor print quality.
Despite variations in retraction settings and other parameters, the stringing issue persists, adversely affecting the overall print quality and structural integrity of the printed objects.

### Expected results

The expected results after following the outlined steps in the Orca Slicer program with a remote extruder 3D printer should include:

Minimal to no stringing occurring between printed parts during the printing process.
Effective retraction settings should retract the filament sufficiently to prevent oozing and stringing between printed sections.
The printed objects should exhibit clean, well-defined surfaces without the presence of unwanted filament strands or artifacts between printed features.
Variations in retraction settings should have a noticeable impact on reducing or eliminating stringing issues, allowing for fine-tuning of parameters to achieve optimal print quality.
Overall, the printed objects should demonstrate improved structural integrity and surface finish, reflecting the successful implementation of appropriate slicing and printing settings to mitigate stringing issues inherent to remote extruder 3D printers.

### Project file & Debug log uploads

[Uploading log.zip…]()

### Checklist of files to include

- [X] Log file
- [X] Project file

Edzi11, 2024-02-16:
Ive been having fuzz and stringing as well that disappears when using prusa slicer. same settings as well.

sixerstyle, 2024-02-29:
Same here, you can't get stringing under control.
no problem with cura or prusa slicer

JimRorie, 2024-04-04:
Ditto on ENDER 6.  I've moved back cura.  No issues with it.  Unsure why these issues can't be tamed.

modaali96, 2024-04-04:
Please note that Changing any settings in the layer hight or first layer hight makes the retraction test fails
layers hight should be 0.2 as the prest settings

vgdh, 2024-04-04:
weird, I don't have this issue with my cheap PETG.

vgdh, 2024-04-04:
attach your project if you can.

closed 2024-07-11 by github-actions[bot]

Rushino, 2024-07-12:
Also suffer stringing.