OrcaSlicer/OrcaSlicer #4012: Not launching after updating Ubuntu

RobRobM, 2024-02-08:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 23.10

### Printer

Bambu Labs X1C

### How to reproduce

1. try to launch Orcaslicer

### Actual results

program crashes after a lot of Gtk-CRITICAL  messages ending with:
(orca-slicer:17595): Gtk-CRITICAL **: 12:17:51.187: gtk_widget_set_size_request: assertion 'width >= -1' failed
1. lo    : 127.0.0.1
2. wlp2s0: ***.***.***.***
3. virbr0: ***.***.***.***
4. enx************: 

### Expected results

Orcaslicer should start.

### Project file & Debug log uploads

[debug_Thu_Feb_08_12_17_48_17595.log.0.txt](https://github.com/SoftFever/OrcaSlicer/files/14208413/debug_Thu_Feb_08_12_17_48_17595.log.0.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

RobRobM, 2024-02-08:
OrcaSlicer_Linux_V1.8.1.AppImage works without issues.

JackBailey, 2024-02-27:
Same here on Fedora 39, 1.9.1 and 1.9.0 don't work, 1.8.1 does

RobRobM, 2024-03-14:
V1.9.1 started working after a an update of the network plug-in. And 2.0.0-beta works as well.

closed 2024-03-14 by RobRobM