OrcaSlicer/OrcaSlicer #4014: Can we get overhangs highlighted in the prepare menue? Some models it is hard to know is supports are needed or not and to what degree. Having the areas that should be are helpful, especially for painting on supports.

poolboy97, 2024-02-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

atl285, 2024-02-08:
In the **view** menu activate **show overhangs**.  Or isn't this what you mean? 😄

kylek29, 2024-02-08:
It's there under support painting, you can then adjust the angle at which they highlight via the slider and use the paint bucket (fill) or other tools to add.

Or do you mean an option to see those highlighted while not in the Support Painting view?

poolboy97, 2024-02-08:
> In the **view** menu activate **show overhangs**.  Or isn't this what you mean? 😄 

Yep. Thats what i was looking for. Sorry guess had a senior moment.

closed 2024-02-08 by poolboy97