OrcaSlicer/OrcaSlicer #4015: PA template calibration model can't be deleted

ohadgur, 2024-02-08:
### OrcaSlicer Version

1.9.0

### OS version

macOS sonoma

### Printer

Ender 3 pro with klipper

### How to reproduce

1. add PA template calibration test
2. delete model placeholder either by clicking and deleting or clicking 'delete all'. also works if adding a new plate and deleting the old one.
3. add another model. I used the handy models, but also primitives work.
4. slice
5. you can see the test is still there

### Actual results

[image]
[image]

### Expected results

when deleting the test object, its shouldn't be sliced

### Project file & Debug log uploads

[PABug.zip](https://github.com/SoftFever/OrcaSlicer/files/14209783/PABug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

LJKraus, 2024-02-13:
See the [Calibration wiki](https://github.com/SoftFever/OrcaSlicer/wiki/Calibration#:~:text=NOTE%3A%20After%20completing%20the%20calibration%20process%2C%20remember%20to%20create%20a%20new%20project%20in%20order%20to%20exit%20the%20calibration%20mode.): 

**NOTE: After completing the calibration process, remember to create a new project in order to exit the calibration mode.**

This applies to all of the calibration tests.

ohadgur, 2024-02-13:
I see, thanks for replying. 
Maybe it's just me, but it wasn't super obvious that that's what I'm supposed to do. I just did it since it didn't work otherwise. perhaps it's worth having some kind of a warning/notification inside the app. 
Another option I can think of is that calibration tests will open a new project/window by default, so it would "feel" like a separate project (maybe even prevent adding other parts to this project, or making it read-only).
If you feel that it's not worth it feel free to ignore me :)

LJKraus, 2024-02-13:
I probably learned about selecting New Project after every test just because I went back to the wiki so many times, learning how to run and interpret the results.

The way all the settings are changed and the test models are loaded, the tests feel like built-in .3mf project files. Maybe they could include instructions in the Project tab, with pictures to help result interpretation, the way that used to be possible before Bambu repurposed it for MakerWorld.  And bring back the editor for our own projects while they are at it!  The whole tab is wasted right now.

closed 2024-05-21 by github-actions[bot]