OrcaSlicer/OrcaSlicer #4016: Objects on slicer plate move after save and reopen

shortbus, 2024-02-08:
### OrcaSlicer Version

1.9.0

### OS version

OS: Mac Orca: 1.9.0

### Additional system information

CPU: Intel

### Printer

Bambu X1C

### How to reproduce

Save, open another Project. Save, open original Project.

### Actual results

all objects on all plate move on the Y axis
[image]

### Expected results

The objects should stay in place between openings. 

### Project file & Debug log uploads

no logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

m2torres97, 2024-05-17:
Still happens even on 2.0.0

joedarrah88, 2024-07-30:
still happens on 2.1.1 and is extremely frustrating

digitalsr2, 2024-08-07:
I have the same issue on my MacBook Pro 16-inch. M1 Max running Sonoma 14.5.

mtbikemarc, 2024-09-12:
Still happening

martincollar, 2024-10-16:
Hi, why this is closed? Im having this issue on both computers (Mac, Win) since i started using OrcaSlicer two months ago. currently happening every day on version 2.2.0-beta, macos 14.6.1 (23G93).. its so annoying, i have projects with maximum plates, and everything needs to be moved again and again. plese fix it...

mtbikemarc, 2024-10-21:
How do we reopen this?

klydex, 2024-11-29:
Still happens on 2.2.0

yomarcyo, 2025-01-04:
@shortbus I re-opend the issue [here](https://github.com/SoftFever/OrcaSlicer/issues/7917) using your screenshot. Hope that's ok for you, as the problem is still unsolved.

closed 2025-01-25 by SoftFever

FelixBuehler, 2025-02-26:
Still not fixed to this day.

mtbikemarc, 2025-03-06:
Can someone please help with this? 

Negative space will also move on it's own after saving/reopening. Not always related to the object moving.

martincollar, 2025-03-07:
its so annoying that devs ignoring such a big bug, im about to leave the slicer due this, i have lot of plates in my projects and everything keeps moving :(

FelixBuehler, 2025-03-07:
It's fixed in the 2.3.0 beta:
https://github.com/SoftFever/OrcaSlicer/pull/7923