OrcaSlicer/OrcaSlicer #4017: Bambu Lab AMS show no color in device tab

Schmudus, 2024-02-08:
### OrcaSlicer Version

2.0.0 (nightly build)

### OS version

Windows 10

### Additional system information

Not necessary

### Printer

Bambu Lab P1S

### How to reproduce

Go to device and load/change filament in AMS.
[image]
[image]
[image]

### Actual results

After loading a filament in device tab the corrsponding slot shows sometimes no color and sometimes no edit button (editing is still working). Looks like a layer shift in the faceplate. PETG seems to be working, PLA matte and PLA HS not. See enclosed Screenshots.

### Expected results

Colored AMS slot

### Project file & Debug log uploads

---

### Checklist of files to include

- [ ] Log file
- [ ] Project file

AngelDLM08, 2024-03-17:
Same issue with colors, but it appears only when selecting custom color.
If selecting standard colors from drop-down menu color renders correctly.

Ruxery, 2024-03-31:
same problem. 
In V1.9.1 it was working. Now in V2.0.0 (beta and release) it is showing no color in the device tab.
[image]

The colors are displayed and can be selected as normal when selecting a print.
[image]

closed 2024-07-07 by github-actions[bot]