OrcaSlicer/OrcaSlicer #4020: Duplicate metadata in gcode might be confusing OctoPrint (and SpoolManager plugin and possibly others)

evilDave, 2024-02-09:
### OrcaSlicer Version

1.9.0

### OS version

MacOS 13.5.2

### Printer

Ender 3 v2 neo

### How to reproduce

1. slice any file
2. send to octoprint with spoolmanager
3. select file in octoprint

### Actual results

1. notice no metadata dropdown
2. notice `Required Filament unknown` and `Cost: no filament from meta`

### Expected results

Should be able to read metadata of gcode file

The code file has this "new" metadata at the top:
```
; total layer number: 116
; filament_density: 1.24
; filament_diameter: 1.75
; max_z_height: 23.20
```

however filament_density and filament_diameter are repeated again after the print instructions. this looks like it might be causing a problem for octoprint when trying to read the metadata (if I remove just these two lines, then octoprint works again). granted, octoprint etc. should be robust against having duplicate lines in a text file, but still... maybe it can be fixed here too... why is this duplicate data written at the top now?

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BrunoBrutus, 2024-03-19:
I have the same issue, also used the 1.9.0 and the latest 2.0.0 Beta.
Removing the filament_density, filament_diameter and the max_z_height solves the problem.
But then I can't send the files directly to the printer and need to pre-process them.
On the 1.8.0 version it's working as expected.

evilDave, 2024-06-07:
Anyone looking at this? I can't guess at why more people are not having this issue? I would have thought it was always an issue for anyone using octoprint with spool manager... maybe that's not that many people...

evilDave, 2024-07-11:
I finally got round to writing a post processing script to repair the gcode, that script is here https://github.com/evilDave/patch-slicer-metadata/blob/main/patch-slicer-metadata.py hopefully that might help someone.

closed 2024-10-18 by github-actions[bot]

nicval, 2024-11-02:
Same problem for me
OctoPrint 1.10.2
OrcaSlicer 2.2.0

ianlini, 2025-04-24:
Confirmed that the issue was resolved in Octoprint 1.11.0.

evilDave, 2025-04-29:
I can also confirm that with latest octoprint the extra comments do not cause a problem.

This can be closed with "don't care, won't fix"

(oh. it was helpfully already closed for being inactive by the bot. I hate that... anyway, not my repo.)