OrcaSlicer/OrcaSlicer #4042: Cyrillic characters are displayed incorrectly in some menus.

AndylgTom, 2024-02-11:
### OrcaSlicer Version

OrcaSlicer V1.9

### OS version

win10

### Additional system information

Cyrillic characters are displayed incorrectly in some menus. In particular in "Embossed text" and Connect to printer.
[image]

### Printer

never mind

### How to reproduce

2

### Actual results

1

### Expected results

2

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Discorez, 2024-02-18:
I have a similar problem.
Orca Slicer versions 1.9.1 and 2.0
Win 11 Pro

Felix14-v2, 2025-09-07:
@SoftFever Could you reopen this issue, please? Still a problem in 2.3.1:

[image]

[image]

[image]

The printer connection window is fixed though.

Felix14-v2, 2025-09-08:
Found that the filament remapping notification also doesn't display cyrillic correctly (the last word is replaced with translation here):
[image]

Felix14-v2, 2025-09-14:
Also there (rotation menu):

[image]

raistlin7447, 2026-08-28:
@Felix14-v2 the "Path to a folder in the model replacing notification" row is already fixed. You reproduced it on ea7117b7 (31 Jul); f444176df8 fixed it a week later as part of #15102.

#15419 covers the other five. The three `?????` rows are the ImGui font atlas, which only loaded Basic Latin for an English UI. The profile-delete dialog and the `Tips:` line are UTF-8 bytes assigned into a `wxString`, decoded with the local code page.

The `Tips:` line was never edited, though. #12155 removed the `#ifndef __WINDOWS__` guard that had kept Windows out of that block, making a pre-existing bug reachable.

closed 2026-08-29 by RF47