OrcaSlicer/OrcaSlicer #4046: Wrong Number of Walls bug: Only-one-wall-top-surfaces to blame !

psiberfunk, 2024-02-11:
### OrcaSlicer Version

1.90 and 2.0.0

### OS version

Win64

### Additional system information

N/A

### Printer

Bambu X1C

### How to reproduce

**Project file** 
[orcaSlicingIssue_WrongWalls.zip](https://github.com/SoftFever/OrcaSlicer/files/14232780/orcaSlicingIssue_WrongWalls.zip)

1. Open project included here and slice.  
2. Observe the resulting slices around layers 73-93 where one section of the model inexplicably has only a single wall instead of 3 like the rest of the model.

**Notes:**

- Mesh is watertight and clean from a solid CAD export.  Verified model geometry is not the issue here to the best of my ability.
- Turning off **Only one wall on top surfaces** off makes it go away!  This seems to be a bug related to this feature.

### Actual results

As noted in the [discord thread here](https://discord.com/channels/1137181739773603922/1206119716251500605) , we are getting a part of the model incorrectly sliced with only 1 wall instead of 3.  This section of the model is in the MIDDLE of the vertical section of the mesh, nowhere near the surface.

**See attached Photos here:**
[image]
[image]

**Actual print result verifies the issue**
[image]

### Expected results

This section of the model should print just like the parts above/below it , with 3 walls !

### Project file & Debug log uploads

[orcaSlicingIssue_WrongWalls.zip](https://github.com/SoftFever/OrcaSlicer/files/14232792/orcaSlicingIssue_WrongWalls.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

psiberfunk, 2024-02-11:
@SoftFever  as we discussed on discord, here is the bug

discip, 2024-02-12:
@psiberfunk
As a workaround set **`One wall threshold`** to **`0`**.

psiberfunk, 2024-02-13:
> @psiberfunk As a workaround set **`One wall threshold`** to **`0`**.

I mean, as a workaround I can just turn off the 1-wall-on-top-surfaces function.. but that's kinda not what i want.  setting the threshold to zero can do other undesirable things .

psiberfunk, 2024-02-19:
> If you turn off the feature you won't have one wall on the top most surface.

Yes, I understand this, and it's helpful for solving my issue with _this print_.  

However, it doesn't solve the general case problem where a user wants both a non-zero threshold AND the feature on, which is a common case. Either the defaults are non-sensible defaults and should be changed so that people won't stub their toes by defaults  OR the feature itself has a glitch where it does unexpected things randomly where clearly even with the threshold at the default 300% of perimeter width doesn't make sense, as the crosssectional area of this non-top region in the slice is huge.

I appreciate the suggestions you're providing as an immediate band-aid to the underlying issue, but the root pathology of something being broken is still there and needs to be fixed.

discip, 2024-02-19:
The only thing I wanted to achieve here is to help you.

I certainly didn't want to talk you out of wanting this problem resolved.

You are absolutely right that this is a bug that needs to be fixed.

psiberfunk, 2024-02-19:
> The only thing I wanted to achieve here is to help you.
> 
> I certainly didn't want to talk you out of wanting this problem resolved.
> 
> You are absolutely right that this is a bug that needs to be fixed.

Sorry, I didn't mean to come off the wrong way.. sometimes people just approach a problem with a "here bandaid it and be happy , but this isn't worth fixing" type of mentality.... And I know _most_ users won't likely be as persistent in discovering which toggle fixes it as I was... certainly I couldn't find it for an hour or so at first.. .

SoftFever, 2024-02-27:
should be fixed with PR #4244

PR #4244 (2024-02-27): Fixed a bug that in edge cases, wrong top surfaces were detected
fixes #4046
closed 2024-02-27 by SoftFever