OrcaSlicer/OrcaSlicer #4047: Default K1 Printers won't show up

MatthaeusLeiner, 2024-02-11:
### OrcaSlicer Version

2.0.0-dev

### OS version

Windows 11

### Printer

Creality K1 Max

### How to reproduce

Install new Version -> Add Printer -> K1 Max, K1 all Nozzles -> Displays after pressing OK only default printer.
After adding K1C it displayed K1 and K1 Max and my Custom K1 Max but no K1C.

### Actual results

[image]
[image]

### Expected results

All printer should show up.

### Project file & Debug log uploads

[Printer Issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14233172/Printer.Issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bistory, 2024-02-11:
You have a conflict with a config named "0.20mm Standard @Creality K1Max (0.4 nozzle)", which is the same name as the K1 Max config of Orca. Remove that configuration and it should fix your issue

MatthaeusLeiner, 2024-02-11:
I removed now all my data from AppData\Roaming\OrcaSlicer\user but I can't add anything. It just shows default.
Or should I search elsewhere?

[image]

bistory, 2024-02-11:
> AppData

The data folder is /Users/YOUR_USER/Library/Application Support/OrcaSlicer

bistory, 2024-02-11:
> I removed now all my data from AppData\Roaming\OrcaSlicer\user but I can't add anything. It just shows default. Or should I search elsewhere? [...]

I don't have Windows to test but you should wipe data folder and re-install Orca

MatthaeusLeiner, 2024-02-11:
The mentioned data folder is mac yes.

I wiped now my data folder and everything called Orca, reinstalled Orca V2.0.0 and it still won't work.
With Orca V1.9.0 it works.

HAndreasH, 2024-02-12:
Same here on latest nightly. Can not create Creality Ender 3.
Seems as if only creality printers are affected.

MatthaeusLeiner, 2024-02-12:
Tried it with the Ender V3 and it won't show up too. I tried it on 2 different laptops.

PR #4072 (2024-02-13): Fix for #4047
It looks like a regression has been introduced in https://github.com/SoftFever/OrcaSlicer/commit/31d1d03578398e5de1f5e173548b5a83e23d7553
"0.20mm Standard @Creality K1Max (0.4 nozzle)" is loaded twice, this PR should fix the issue.

And then fix #4047
closed 2024-02-13 by SoftFever