OrcaSlicer/OrcaSlicer #4062: Bad stringing that cant be tuned out.

Glenno-H, 2024-02-13:
### OrcaSlicer Version

1.9.0 and 2.0.0

### OS version

Windows 11

### Additional system information

13900k
32gb Ram

and 

8700
16gb Ram

### Printer

Voron 2.4 350mm - SB with CW2 and Dragon

### How to reproduce

From the Orca Slicer Calibration menu, choose Retraction test, the values set dont really seem to matter.

I tried the suggestions here

https://www.reddit.com/r/FixMyPrint/comments/1764ewy/stringing_after_switching_to_orca_slicer/

With the only one helping was to reduce the "Z Hop when retracting" to 0mm (But I dont want to disable Z Hop!)

### Actual results

Stringing regardless of retraction settings.  Not even any visable improvement as its prints as increases retraction distance.

### Expected results

Less stringing as the retraction calibration print increases retraction distance.  The Gcode does show that retraction distances are being increased as the print progresses.

### Project file & Debug log uploads

[debug_Tue_Feb_13_09_35_04_15948.log](https://github.com/SoftFever/OrcaSlicer/files/14253913/debug_Tue_Feb_13_09_35_04_15948.log)
[Retraction test.zip](https://github.com/SoftFever/OrcaSlicer/files/14253919/Retraction.test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Edzi11, 2024-02-16:
I have been struggling with this for the last 3 months. prusa slicer doesnt give me any stringing or fuzz but orca does.

Santoshsomi, 2024-02-19:
Tried changing Z hop type to different ones? and reduce the zhop distance to min 0.2mm?

Edzi11, 2024-02-19:
Ya Ive tried everything. I've been trying to solve this issue for 3 months straight with no solution other than recently trying prusa and super slicer. Those ones don't give me any stringing and the settings are as close to the same as I can get them.

matthewj301, 2024-02-22:
Same issue here, especially when printing multiple large objects - its like it doesn't actually retract at the end of the layer before the travel move. On top of that, retraction test comes out perfectly at 0.15mm, but not when doing an actual print :) They mentioned fixing something relating to retraction in yesterday's 1.9.1 release, so I am running a test print now.

matthewj301, 2024-03-07:
Still an issue, setting min travvel distance for retractions to 0 helped, but we should't have to do that

grrminator, 2024-04-18:
I know i commented on another one of these but my github profile shows it just .... disappeared. Odd.

But same. Coming from cura to orcaslicer. Cura does not have this stringing issue. But Orcacliser does with the exact same settings or better settings.

@SoftFever is there any fix in the works? this githubs issue history is long with stringing issues and if you google orcaslicer stringing a lot of people have this problem when coming from other slicers.

closed 2024-07-25 by github-actions[bot]