OrcaSlicer/OrcaSlicer #4066: Currently it is possible to enable infill combination but not to specify how many layers we want the infill combined by which is the case on superslicer

lewiss66, 2024-02-13:
### OrcaSlicer Version

1.9.0

### OS version

windows 11

### Additional system information

Currently it is possible to enable infill combination but not to specify how many layers we want the infill combined by.

When printing at small layer heights like 0.1mm I have found that combining infill can drastically speed up print times. But by default the combined infill prints a thick infill only every 4 layers at 0.4mm thick. This has made for some poor top surface finish on some of my parts.

The above example sliced in superslicer it is possible to specify to print infill every 2nd layer for instance resulting in a 0.2mm thick infill or even every 3rd layer.

I feel that this would be benifical to specify printing every x number of layers. 1 being the default setting where combined infill is disabled

### Printer

ender 3

### How to reproduce

Currently it is possible to enable infill combination but not to specify how many layers we want the infill combined by.

### Actual results

When printing at small layer heights like 0.1mm I have found that combining infill can drastically speed up print times. But by default the combined infill prints a thick infill only every 4 layers at 0.4mm thick. This has made for some poor top surface finish on some of my parts.

### Expected results

Like in superslicer it is possible to specify to print infill every 2nd layer for instance resulting in a 0.2mm thick infill or even every 3rd layer.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

weekendkoder, 2024-03-28:
Quite usefull. Would be great to have it soon

Nuc1eoN, 2024-06-27:
I support this case

ZedOud, 2024-07-26:
These may be relevant: #2757 #2735 #2175

igiannakas, 2024-08-09:
Raised a PR that aims to address this above.

closed 2024-09-07 by SoftFever