OrcaSlicer/OrcaSlicer #4067: Add an option to allow to blacklist speed ranges to avoid certain speeds to compensate VFA.

NeverUsedID, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
Faster printers often have problems with VFA on certain speeds

**Which printers will be beneficial to this feature**
Bambu Lab Printers, Crealiyt K1 Series. Proberly all high speed printers,

**Describe the solution you'd like**
Add an option to blacklist certain speed ranges, while using features like "Slow printing down for better layer cooling" or "Slow down on overhangs"

Example: Printer has higest VFA on 100mm/s, so i would enable blacklist for 90-110ms/s. The Slicer decides to use 89mm/s for a layer which would be otherwise printed at 100mm/s because of "Slow printing down for better layer cooling"

czepter, 2024-02-26:
Would love to see this implemented! Would reduce unwanted vibrations / noise at certain speeds as well :-)

KaruroChori, 2024-05-06:
I also would like to see this implemented. For example, this is the resonance graph directly linked of a flashforge adventurer 5m:
[image]
it would be nice to exclude speeds around 50, 100 and possibly higher order peaks as well.

rodrigo2019, 2024-05-09:
I think there is another point to be concerned. Let's assume that we have a VFA on axis X at 100mm/s. If the toolhead is moving at 141mm/s at diagonal (45º) it means that your X will move at approximately 100mm/s, the same speed that you want to avoid.

VFA are purely generated by a specific speed in specific axis? Or it can be generated by the combination of the axes?

KaruroChori, 2024-05-10:
I implemented this idea directly in a printer firmware I am working on.

The approach for optimization taken is a very simplistic greedy one, but it should operate well enough in most realistic scenarios (and if optimized it is computationally cheap). It is extremely easy to adapt it to corexy, less so to deltas I guess.
So far it takes into account constraints for the extruder, x and y.

It should be tested and improved with some additional fail-safe mechanisms, but I am leaving it here as reference
```
const forbidden_x: [number, number][] = [
    [Number.NEGATIVE_INFINITY, 0],
    [45, 65],
    [85, 115],
    [190, 210],
    [500, Number.POSITIVE_INFINITY]
];
const forbidden_y: [number, number][] = [
    [Number.NEGATIVE_INFINITY, 0],
    [45, 65],
    [85, 115],
    [190, 210],
    [500, Number.POSITIVE_INFINITY]
];

const forbidden_e: [number, number][] = [
    [Number.NEGATIVE_INFINITY, 0],
    [25, Number.POSITIVE_INFINITY]
]

function closest(current: number, interval: [number, number], eps: number = 10e-4) {
    return (current - interval[0] < interval[1] - current) ? interval[0] - eps : interval[1] + eps
}

//Assuming a Cartesian kinematic.
function move(
    from: [number, number],
    to: [number, number],
    e: number,
    v: number,
    MAX_STEPS: number
) {
    const dx = to[0] - from[0];
    const dy = to[1] - from[1];
    const dw = Math.sqrt(Math.pow(dx, 2) + Math.pow(dy, 2));
    const dt = dw / v;

    const vx = dx / dt;
    const vy = dy / dt;

    let vx_c = vx;
    let vy_c = vy;
    let e_c = e

    let steps = 0
    for (; steps < MAX_STEPS; steps++) {
        //Adapted flow rate target
        e_c = e * Math.sqrt(Math.pow(vx_c, 2) + Math.pow(vy_c, 2)) / v;

        //Bad performance just for the sake of some simpler code
        const check_x = forbidden_x.filter((x) => vx_c < x[1] && vx_c > x[0]);
        const check_y = forbidden_y.filter((y) => vy_c < y[1] && vy_c > y[0]);
        const check_e = forbidden_e.filter((e) => e_c < e[1] && e_c > e[0]);

        if (check_x.length === 0 && check_y.length === 0 && check_y.length === 0) break;

        //If flow is wrong first adapt it
        if (check_e.length !== 0) {
            const factor = closest(e_c, check_e[0]) / e_c
            e_c *= factor
            vx_c *= factor
            vy_c *= factor
        }

        else if (check_x.length !== 0) {
            const factor = closest(vx_c, check_x[0]) / vx_c
            e_c *= factor
            vx_c *= factor
            vy_c *= factor
        }

        else if (check_y.length !== 0) {
            const factor = closest(vy_c, check_y[0]) / vy_c
            e_c *= factor
            vx_c *= factor
            vy_c *= factor
        }
    }
    if (steps === MAX_STEPS) {
        console.error(`Unable to find a suitable solution in ${MAX_STEPS} steps`)
    }
    console.log(steps)

    console.log("Initial: ", from, to, e, v, vx, vy)
    console.log("Final: ", from, to, e_c, Math.sqrt(Math.pow(vx_c, 2) + Math.pow(vy_c, 2)), vx_c, vy_c)

    /*
    queue.pushTransaction([
        ["move_x", { v: vx_c, from: from[0], to: to[0] }],
        ["move_y", { v: vy_c, from: from[1], to: to[1] }],
        ["extrude", { e: e_c }],
    ]);
    */
}

//Test this move
move([0, 0], [200, 100], 30, 130, 10)
```

KaruroChori, 2024-05-10:
I made some further evaluations about the delta kinematic. It seems like this feature could only be easily implemented for the z axis.

Speed filtering for a delta kinematic would be harder to implement on the xy plane, because linear movements in Cartesian coordinates would be strongly non linear in the "motor space".
This would make most of the simplest optimization strategies not feasible for longer moves from the center to the edge of the printable area.

Mathematically it can be resolved, but it would likely introduce flow artifacts on straight lines, unless we have an excellent flow calibration implemented for our printer. Assuming this can be sorted out, at least in principle, speed filtering would also work for deltas.

closed 2024-08-17 by github-actions[bot]