OrcaSlicer/OrcaSlicer #4068: Change wall order only on overhangs

cosminr86, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

improve overhangs quality and overall print quality

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

all printers

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

As orca can detect overhangs, I think it will be beneficial to use wall orderin inside to outside where the overhangs are, and use outside to inside where is not needed

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

A wall ordering option on overhangs will be wonderfull

Thank you!

killerline10168, 2024-02-15:
+1

I also had the same idea a couple of weeks ago, but it felt like the devs had their hands full. This feels like a somewhat simple yet great solution to bad overhangs. We have to choose between layer consistency or nice overhangs, but why not both? :)

jet689, 2024-03-07:
I 3rd this idea

closed 2024-06-13 by github-actions[bot]

eternal-sorrow, 2025-02-19:
Please reopen this. Automatic issue closing shouldn't be a thing.

imagesurgery, 2025-03-17:
- I can only get rid of seams completely when using I/O/I or O/I
- I can only get good overhangs with I/O

This is a feature I would love!

tjeerdvandijk, 2025-04-09:
This would be a fantastic feature to ensure quality walls when printing a model with overhangs.